ArduPilot/ardupilot #4305: Servo output not working on pixhawk

kudykam, 2016-06-12:
#### Issue details
I bought Tower PRO MG90S (digital servo) and tried to connect to pixhawk. Servo Rail is powered via bec
1. Connected  servo to RC7 and set RC7 to passthrough. But no response  from servo. But status shows ch7out changing value when turning knob
2. Tried gimbal tilt function (optional hardware) on RC9  (connect servo to AUX1) and input channel set to ch7(to control tilt from radio as well). But this also doesnt react on quad movement or radio input.
3. Servo connected directly to channel 7 on TX works. Rádio and TX is from Cheerson cx20

#### Version
3.4 den, didn't tried lower Version 

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
Pixhawk 2.4.8 

#### Logs
Dont have any

rmackay9, 2016-06-13:
Remember to press the safety switch.  In Copter-3.4 the servos don't move unless the safety switch is pushed first.

rmackay9, 2016-06-13:
By the way, support should go into discourse so I'm going to close this.
http://discuss.ardupilot.org/c/arducopter/copter34

closed 2016-06-13 by rmackay9