ArduPilot/ardupilot #4311: Don't update home at 10Hz

WickedShell, 2016-06-14:
#### Issue details
There is not a significant benefit to updating home at 10Hz, and it causes a continuous stream of new HOME_POSITION messages to be sent out the radio. The core reason this is frustrating me is that from the GCS side I'm using this message to kick off some relatively intensive work (optimization there is easily possible), and that this is eating a significant chunk of radio bandwidth (by my pencil and paper calculations its ~10% of the link budget for the default 57600 baud).

Are there any objections to moving this to the 1Hz loop rather then the 10Hz GPS loop?

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

rmackay9, 2016-06-14:
This is a Plane issue right?  Copter only updates the home position when it first gets a good EKF position estimate and then when it arms.

WickedShell, 2016-06-14:
It's plane and rover, I wasn't sure if copter was doing that as well or not
(it at least didn't have the same copy pasted update_home function, but I
wasn't sure if it had a different one.). I'll retag it as plane and rover
only.

magicrub, 2016-06-14:
I agree that tagging home at 10 Hz is overkill, I didn't realize that it was generating that GCS message. Makes sense though. I don't see any downside to at least reducing it to 1 Hz. Changing it to setting once on initial GPS lock and again when armed is something we could discuss.

WickedShell, 2016-06-14:
It's technically in the 10Hz loop and when you have a new GPS update. But I'm flying a GPS at the moment that is providing data at 10Hz, and will be submitting a u-blox patch to enable higher rates there. Otherwise you will typically only see it at 5Hz with the normal u-blox driver.

magicrub, 2016-06-14:
I've never liked the continuos home-setting logic living in the GPS driver. Should be event driven or 1Hz

tridge, 2016-10-16:
fixed

closed 2016-10-16 by tridge