ArduPilot/ardupilot #4314: RCn_TRIM not working AP 3.6.0

Autonomost, 2016-06-14:
#### Issue details
When I am trying to trim the RCn_ ouputs it will only trim one of the two servos. Even selecting different servos within mission planner it still will not trim the correct one. 

#### Version
AP 3.6.0 and AP 3.5.0 at 0.75

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Airframe type
Plane with Two Elevators, Rudder, Two Ailerons, Two Flaps. Supercub 

#### Hardware type
Pixhawk

#### Logs
#NOTE: 6/14/2016 6:22:12 AM Plane: Skywalker
ACRO_LOCKING,0
ACRO_PITCH_RATE,180
ACRO_ROLL_RATE,180
ADSB_BEHAVIOR,0
ADSB_ENABLE,0
AFS_AMSL_ERR_GPS,-1
AFS_AMSL_LIMIT,0
AFS_DUAL_LOSS,1
AFS_ENABLE,0
AFS_GEOFENCE,1
AFS_HB_PIN,-1
AFS_MAN_PIN,-1
AFS_MAX_COM_LOSS,0
AFS_MAX_GPS_LOSS,0
AFS_QNH_PRESSURE,0
AFS_RC,1
AFS_RC_FAIL_TIME,0
AFS_RC_MAN_ONLY,1
AFS_TERM_ACTION,0
AFS_TERM_PIN,-1
AFS_TERMINATE,0
AFS_WP_COMMS,0
AFS_WP_GPS_LOSS,0
AHRS_COMP_BETA,0.1
AHRS_EKF_TYPE,2
AHRS_GPS_GAIN,1
AHRS_GPS_MINSATS,6
AHRS_GPS_USE,1
AHRS_ORIENTATION,0
AHRS_RP_P,0.2
AHRS_TRIM_X,0
AHRS_TRIM_Y,0
AHRS_TRIM_Z,0
AHRS_WIND_MAX,0
AHRS_YAW_P,0.2
ALT_CTRL_ALG,0
ALT_HOLD_FBWCM,0
ALT_HOLD_RTL,10000
ALT_MIX,1
ALT_OFFSET,0
ARMING_ACCTHRESH,0.75
ARMING_CHECK,1
ARMING_MIN_VOLT,0
ARMING_MIN_VOLT2,0
ARMING_REQUIRE,1
ARMING_RUDDER,1
ARSPD_AUTOCAL,0
ARSPD_ENABLE,1
ARSPD_FBW_MAX,22
[... 592 lines cut ...]
TKOFF_THR_MINSPD,0
TKOFF_THR_SLEW,0
TRIM_ARSPD_CM,1200
TRIM_AUTO,0
TRIM_PITCH_CD,0
TRIM_RC_AT_START,0
TRIM_THROTTLE,45
TUNE_CHAN,0
TUNE_CHAN_MAX,2000
TUNE_CHAN_MIN,1000
TUNE_ERR_THRESH,0.15
TUNE_MODE_REVERT,1
TUNE_PARAM,0
TUNE_RANGE,2
TUNE_SELECTOR,0
USE_REV_THRUST,2
VTAIL_OUTPUT,0
WP_LOITER_RAD,60
WP_MAX_RADIUS,0
WP_RADIUS,90
YAW2SRV_DAMP,0
YAW2SRV_IMAX,1500
YAW2SRV_INT,0
YAW2SRV_RLL,1
YAW2SRV_SLIP,0

gmorph, 2016-06-15:
This isn't the place for support issues - please raise this request in the support forums.
http://discuss.ardupilot.org/
Thanks.

closed 2016-06-15 by gmorph

Autonomost, 2016-06-15:
I was just about to take it down. Sorry. 

Evan Hammac