ArduPilot/ardupilot #4316: install-prereqs-ubuntu.sh missing libc6-i386

dricketts, 2016-06-14:
#### Issue details
The script install-prereqs-ubuntu.sh seems to be missing libc6-i386. This results in error messages of the form:

   opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-gcc: No such file or directory

when running ```make px4-v2```. According to @peterbarker, the solution is probably as simple as adding ```libc6-i386``` to the ```PX4_PKGS``` list in that file

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[ - ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
Pixhawk

#### Logs
N/A

magicrub, 2016-06-14:
problem confirmed

closed 2016-06-16 by magicrub

magicrub, 2016-06-16:
Fixed, thanks @dricketts and @peterbarker