ArduPilot/ardupilot #4317: AP should stop using the elevator for atitude/altitude control for roll between 90 and 120 degrees (or even 70 to 120 degrees)

marcmerlin, 2016-06-15:
https://github.com/ArduPilot/ardupilot/issues/3875 shows that currently AP will use the elevator when the plane is inverted and it knows to invert the elevator position.
However, for a roll of 90 to maybe up to 135 degrees, the elevator does little to very little to control pitch and altitude, and using it reversed causes some backwards spin that stalls both wings and prevents the ailerons from turning the plane right side up

In https://github.com/ArduPilot/ardupilot/issues/3925 I proposed a technically superior handling of this problem (mixing rudder and elevator depending on roll angle), but since it may take longer to happen, and the current behaviour badly crashed my plane twice, I'm proposing a simpler stop gap measure. To choose between:
1) do not use elevator for pitch between rolls of 90 and 120 (most conservative)
2) between 70 and 120 (adding 70 to 90 doesn't fix crash problems, but stops giving input equivalent to a lot of rudder when you're trying to command pitch)
3) you could go all the way to 135 degrees instead of 120, but I'm not going to push for that. Still, does an auto mode have any business flying inverted and past a 45 degree angle? If you think so, I at least want to push for not allowing past 60 degrees and if you go past that, no more use of the elevator

magicrub, 2016-06-15:
Hi @marcmerlin We discussed this at our last dev call last, its definitely a problem and the proper way to handle it is to mix in rudder given the detected roll.

marcmerlin, 2017-06-06:
Was this more or less fixed in 3.6 or 3.7? I think @tridge took care of that one, right?
If so, feel free to close.

marcmerlin, 2017-06-06:
#3925 would still be very cool, but not a safety issue