ArduPilot/ardupilot #4324: GLOBAL_POSOITION_INT.relative_alt to use terrain+lidar data

magicrub, 2016-06-16:
GLOBAL_POSOITION_INT.relative_alt to use terrain+lidar data instead of gps.alt - gps.home.

For plane, simply use [relative_ground_altitude()](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/altitude.cpp#L140) instead of [relative_altitude](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/altitude.cpp#L123)(). That function is not on copter but can easily be added.

Any objections?

WickedShell, 2016-06-16:
It is a poor choice of you have a mixture of relative and terrain alt
frames in the same mission... we still need the other version for any
waypoints that are only relative. I'd rather see mavlink 2 and the packet
extension address this.

magicrub, 2016-06-16:
I don't understand your argument. What is 

> mixture of relative and terrain alt
> frames in the same mission

 and why does that matter? How is knowing a more accurate relative alt via terrain+rangefinder a problem for relative waypoints? This ends up allowing the GCS know where the ground is when before you just don't but the plane knows it.

WickedShell, 2016-06-16:
I can have a mission where waypoint 1-5 are all relative altitude (not terrain following) while waypoints 6-29 are all terrain following waypoints. Under your proposed changes I don't see how I would have an appropriate altitude being reported when trying to fly waypoints 1-5. The reason the terrain+altitude is a problem here is that telling the GCS that is fine, but that's not what the autopilot is flying from.

magicrub, 2016-06-16:
Suggestion from @WickedShell via mumble: Implement and use https://pixhawk.ethz.ch/mavlink/#ALTITUDE

magicrub, 2016-06-16:
Next question is, if there's this new packet to stream what stream do we put it in?

magicrub, 2016-06-16:
thinking of adding it to SRx_EXTRA1

magicrub, 2016-06-17:
closed in favor of sending rangefinder data in terrain packet in issue https://github.com/ArduPilot/ardupilot/issues/4328

closed 2016-06-17 by magicrub