ArduPilot/ardupilot #4331: Safety switch does not disarm differential spoilers

raptortrias, 2016-06-17:
#### The safety switch does not disarm differential spoilers

#### 2.6.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Airframe type
Flying wing, ELEVON_OUTPUT,

#### Hardware type
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2016-06-17:
But it does disarm the other channels? What channels were your diff
spoilers on?

raptortrias, 2016-06-17:
@magicrub 

On power up all channels are disarmed, once safety switch is on , all servo channels are armed, then if I disarm using safety switch the dif spoilers remain on, the other ailerons are disarmed.

 ELEVONS are on RC1 and RC2, spoilers are on AUX1 (RC9) and AUX2 (RC10).

 Was this because they are on AUX ports?

 Thanks

magicrub, 2016-06-17:
Ahh yes, I think the saftey switch only applies to channels 1-8.

raptortrias, 2016-06-17:
@magicrub 

My wrong,

 thanks

closed 2016-06-17 by raptortrias

magicrub, 2016-06-17:
That is a hardware limitation, not software, so it should work the same on
all versions. however, there were some saftey-button and diff spoiler
related changes in the latest release. Is the behavior different than
previous fw versions?

WickedShell, 2016-06-17:
@magicrub On plane 3.6 safety switch applies to FMU channels as well, which
is why people have had problems with servo gimbals/servo camera triggers

magicrub, 2016-06-17:
So we should reopen this? @WickedShell please advise

WickedShell, 2016-06-17:
I suspect so, as I don't think the safety switch explains this on 3.6. But
the initial post said 2.6.0 so I didn't know what to make of that....
@tridge thoughts?

raptortrias, 2016-06-17:
@WickedShell  apologies I have corrected the FW version to 3.6.0