ArduPilot/ardupilot #4333: AC3.4 rc1 optical flow issue with lidar lite

roque-canales, 2016-06-17:
Hello,

I'have made new tests with lidar lite and optical flow, and I have issued many burst on Z axis.

I was testing the system in indoor on a textured concrete floor types.

XY positioning was perfect.

how to weigh less lidar lite to avoid these outbursts on the Z axis?

Also there is possible to know details about ek2 to better set gates for lidar lite and optical flow to reject bad readings ?

magicrub, 2016-06-17:
I've heard reports of the lidar's laser being seen by the optical flow
sensor. Try putting an IR filer over the optical flow.

rmackay9, 2016-06-18:
We're handling the Copter-3.4 support over in the forums so could you re-raise it over there?  Including a dataflash log would also be very helpful.
http://discuss.ardupilot.org/c/arducopter/copter34

closed 2016-06-18 by rmackay9