ArduPilot/ardupilot #4337: APMrover2 SITL : RC override don't work in Guided

khancyr, 2016-06-18:
#### Issue details
Sending RC override in rover SITL in GUIDED mode doesn't work. Command are received but the rover doesn't follow it...
But it works in Manual mode.

#### Version
Master SITL 
Need test on real rover

#### Platform

[ x] Rover

#### How to reproduce 

Launch Rover SITL in skid steer or not. 
set Mode Guided
ARM
Send Rc override .

magicrub, 2016-06-18:
Is it supposed to work in guided? There's guided commands to control it in
guided...

khancyr, 2016-06-19:
It should be working.  It is the simplest way to control rover with companion computer. It works in Copter. 
The guided controls are pretty poor on rover since we only handle Gps like guided waypoint and don't have Setpoint command

squilter, 2016-06-20:
Guided mode does not respect RC input, and that is by design. Here is the wiki page for guided mode:
http://ardupilot.com/rover/docs/guided-mode.html

closed 2016-06-20 by squilter