ArduPilot/ardupilot #4338: Error compiling ArduPlane SITL in Mac OS X Yosemite

alejamp, 2016-06-18:
#### Issue details
When I executed **make sitl**, got a bunch of compilation errors. Here are some of then:
Missing definitions, errors such as "feenableexcept" is not implemented. 
Missing declaration CLOCK_REALTIME, there is a problem with get_realtime. 
Error at linking time: ld: library not found for -lrt

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover

#### Airframe type
-

#### Hardware type
SITL

magicrub, 2016-06-18:
Can you post the compiler output?

alejamp, 2016-06-18:
Got it fixed.
Missing **CLOCK_REALTIME**, there is a problem with get_realtime in Mac OS X. I need to use an alternative method which I picked from: https://gist.github.com/jbenet/1087739, there is a workaround.
Missing **feenableexcept** fixed skiping it using __APPLE__ flag.
Cherry pick this commit https://github.com/alejamp/ardupilot/commit/f5bb0d8b68f683329ba64fcbabb98c1e0851b511

In order to solve linking issue at compilation: **ld: library not found for -lrt** 
modify mk/board_native.mk at line 65. Remove/comment -lrt and -pthread params.

magicrub, 2016-06-18:
Pull requests welcome!

bugobliterator, 2016-06-23:
This PR https://github.com/ArduPilot/ardupilot/pull/4350 should resolve the issue

peterbarker, 2018-03-01:
People are building successfully now.

closed 2018-03-01 by peterbarker