ArduPilot/ardupilot #4340: Add number of deferred messages to SYS_STATUS errors_count1

WickedShell, 2016-06-19:
#### Issue details
https://pixhawk.ethz.ch/mavlink/#SYS_STATUS specifies that errors_count 1-4 are autopilot specific error counters. I'd like to propose that we use errors_count1 to report the number of deferred MAVLink messages in the queue. 

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

rmackay9, 2016-06-21:
Seems like a good health check.  I think we should also log it in the PM dataflash message.

WickedShell, 2018-08-26:
Doesn't seem particularly likely to make it in, and the stream rate scaling we do means that this automatically gets balanced out till it's at 0 in flight, without any good indication to the GCS this error might spike but will always trend towards 0 without any real indication why. Closing the idea for now as not worthwhile.

closed 2018-08-26 by WickedShell