ArduPilot/ardupilot #4341: The problem  about EKF and Attitude Algorithm in ardupilot-Copter-3.3

Hertz11, 2016-06-19:
Hello,
I found that EKF will go wrong when the quadcopter in high speed flight without GPS . I used ardupilot-Copter-3.3-rc5 on PixHawk.It shows "pos vert variance" in Mission Planner 1.3.32.I put "hgtVar" of "getVariances(velVar, posVar, hgtVar, magVar, tasVar, offset);" out,I found it's incredible,sometimes it over than 8.0 .However,it normally less than 0.02 .Another problem is the attitude algorithm can't work well in high speed(such as 10m/s).

The conditions can be list as follows.

1.Remove the GPS and keep the other sensors(accelerometer/gyroscope/barometer/external compass) on quadrotor,set the flight mode to "stabilize",after take off from ground about 50 meters,play the full pitch joystick to go ahead. When the plane fly away from the home more than 200 meters, set the flight mode to "althold" immediately,the plane will triumph in the skies,it's out of control.It's important to note,sometimes the speed will more than 20(or 25) meters per second.

2.In addition,remove the GPS and keep the other sensors(accelerometer /gyroscope /barometer /external compass) on quadrotor,set the flight mode to "althold",after take off from ground about 50 meters,play the full pitch joystick to go ahead. When the plane fly away from the home more than 200 meters,the plane will triumph in the skies too.

3.If the GPS is valid ,the flight will be normal in the above case.So I guess the problem is caused by height estimation without GPS.

4.set the flight mode to "stabilize".After take off from ground about 50 meters,play the pitch joystick to go ahead and keep the speed about 10m/s. When the plane fly away from the home more than 200 meters, play the pitch joystick to go back until the plane close to home.Then repeat such operations for several times, you will find you can't correct the attitude of the plane by joystick,the attitude angle is bigger and bigger,in worst case, it may be over 80 degrees.

As you see, my English is poor, please forgive me.

Really need your help!
Thank you!

#### Platform
Copter

#### Airframe type
x_quadrotor

#### Hardware type
pixhawk

magicrub, 2016-06-19:
Ekf has been improved, please test again using v3.4

OXINARF, 2016-06-19:
@Hertz11 why are you using Copter 3.3-rc5 when the oficial release is at Copter 3.3.3 and we are beta-testing Copter 3.4-rc1?
Also, this is a support question - and not a bug in current implementation - and support questions should be addressed in the forum (http://discuss.ardupilot.org). For that reason I'm closing this. Please reopen (or open a new one) if you have this issue with Copter 3.4-rc1 or master.

closed 2016-06-19 by OXINARF