ArduPilot/ardupilot #4349: YAW reverse Differential Spoilers Flying wing

raptortrias, 2016-06-21:
#### Reversed YAW 

Differential spoilers (@ AUX9 and AUX10) have been setup on Flying wing .

 Can obtain correct YAW movement with RC, however autopilot recognizes opposite YAW movement (arming requires full left) and YAW correction in FBWA is opposite to what it should be.

 Have tried reversing YAW on RC and RC4_REV but it won't fix the problem. 

PS Probably doing something silly here myself (?)

#### Version
Plane 3.6.0
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover

#### Airframe type
Flying wing

#### Hardware type
pixhawk

#### Logs
NA

magicrub, 2016-06-21:
Please post your param file

raptortrias, 2016-06-21:
@magicrub  Either rudder arm would work correctly or RC YAW

[FW_21_6_2016.zip](https://github.com/ArduPilot/ardupilot/files/325164/FW_21_6_2016.zip)

Thanks

DavidIngraham, 2018-05-04:
Closing, as this appears to be a setup issue. Also, not relevant after the Servo Split

closed 2018-05-04 by DavidIngraham