ArduPilot/ardupilot #4355: SITL Killing sim_vehicle.py doesn't terminate the mavproxy instance

WickedShell, 2016-06-22:
#### Issue details
Terminating sim_vehicle.py from the console with ctrl-c leaves the mavproxy instance running.

#### Version
ba2382c

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

lucasdemarchi, 2016-07-11:
@WickedShell do you know if there is a good way on windows to track the process we create? Does sim_vehicle.sh work fine?

WickedShell, 2016-07-12:
I don't know a good way. sim_vehicle.sh can't be ctrl-c'd for me, the only way to get the terminal back is to actually externally close/kill mavproxy first. This is somewhat better in the sense of I can't leave the port locked by accident and not realize that SITL is still running in the background.

WickedShell, 2016-07-12:
I did testing in the past about killing the cygstart process but I can't recall if that actually worked....

lucasdemarchi, 2016-07-12:
Also... do you have psutil installed and it didn't work or you don't have psutil?  I think we were using psutil before but then switched to kill_tasks_cygwin() we have now due to it not working on windows.  However, looking a little bit around it looks like psutil is available there.

WickedShell, 2016-07-12:
psutil is not available in cygwin for me.

I tried to find a way to get that installed before and it turned into a
rabbit hole of impossibility/no success. Spent about an hour and a half
chasing internet instructions and never got it to work.

khancyr, 2018-04-24:
I think this as been solve. 
Reopen if I am wrong

closed 2018-04-24 by khancyr