ArduPilot/ardupilot #4358: Plane: Log target loc to NTUN

WickedShell, 2016-06-22:
#### Issue details
When trying to diagnose controlled fly aways or navigational errors from DF logs it would be valuable to know where the plane was attempting to fly.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

enjoybeta, 2018-03-22:
why not adding into tlogs if you lost the plane?

enjoybeta, 2018-03-22:
I think it's done. WpDist, TargBrg, and NavBrg can be found in NTUN, this issue can be closed

WickedShell, 2018-03-22:
WpDist, TargBrg, NavBrg aren't sufficient you really need to know lat/lon/alt.

enjoybeta, 2018-03-24:
You mean the target lat/lon/alt? I don't think it's avilable for all modes. For what mode can we get target lat/lon/alt? AUTO? Any more?

WickedShell, 2018-03-24:
It's available for rtl, guided, cruise, loiter at a minimum.

enjoybeta, 2018-03-25:
Do you know which funtion logs the data for NTUN? Is is inside DataFlash class? I am trying to work on it.

PR #8492 (2018-06-11): Plane: Log navigation target location
Closes #4358
commit fae8d1e4 (2018-06-11): Plane: Log navigation target location

Closes #4358
closed 2018-06-11 by tridge