ArduPilot/ardupilot #4359: Unable to make SITL for arduplane

RoverHunter, 2016-06-22:
#### Issue details
After installing Cygwin with the required packages, I am unable to make sitl for plane. I can compile the code using eclipse but not on cygwin. This problem is just present on ArduPlane. ArduCopter is just fine.

O get the following error:
/cygdrive/c/ardupilot/arduplane
$ make sitl
Checking modules
// BUILDROOT=/tmp/arduplane.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="a4277680" -I/cygdrive/c/ardupilot/libraries/AP_Common/missing -DMAVLINK_PROTOCOL_VERSION=2 -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE
%% tuning.o
<command-line>:0:21: warning: "APM_BUILD_arduplane" is not defined [-Wundef]
/cygdrive/c/ardupilot/libraries/AP_Vehicle/AP_Vehicle_Type.h:36:41: note: in expansion of macro 'APM_BUILD_DIRECTORY'
 #define APM_BUILD_TYPE(type) ((type) == APM_BUILD_DIRECTORY)
                                         ^
/cygdrive/c/ardupilot/libraries/AP_BattMonitor/AP_BattMonitor.h:109:5: note: in expansion of macro 'APM_BUILD_TYPE'
 #if APM_BUILD_TYPE(APM_BUILD_ArduPlane)
     ^
<command-line>:0:21: warning: "APM_BUILD_arduplane" is not defined [-Wundef]
/cygdrive/c/ardupilot/libraries/AP_Vehicle/AP_Vehicle_Type.h:36:41: note: in expansion of macro 'APM_BUILD_DIRECTORY'
 #define APM_BUILD_TYPE(type) ((type) == APM_BUILD_DIRECTORY)
                                         ^
/cygdrive/c/ardupilot/libraries/AP_BattMonitor/AP_BattMonitor.h:130:5: note: in expansion of macro 'APM_BUILD_TYPE'
 #if APM_BUILD_TYPE(APM_BUILD_ArduPlane)
     ^
cc1plus: warning: unrecognized command line option "-Wno-unknown-warning-option"
In file included from /cygdrive/c/ardupilot/libraries/AP_Math/AP_Math.h:16:0,
                 from /cygdrive/c/ardupilot/libraries/AP_GPS/AP_GPS.h:22,
                 from Plane.h:38,
                 from tuning.cpp:3:
/cygdrive/c/ardupilot/libraries/AP_Math/quaternion.h: In member function 'bool Quaternion::is_nan() const':
/cygdrive/c/ardupilot/libraries/AP_Math/quaternion.h:56:24: error: 'isnan' was not declared in this scope
         return isnan(q1) || isnan(q2) || isnan(q3) || isnan(q4);
                        ^
compilation terminated due to -Wfatal-errors.
C:/ardupilot/mk/build_rules.mk:11: recipe for target '/tmp/arduplane.build/tuning.o' failed
make: *** [/tmp/arduplane.build/tuning.o] Error 1

Please note that I can make sitl for version 3.4. I did not try any other releases.

Am I missing something?

#### Version
Official 3.6.0 released 2016-06-11 and the latest released 2016-06-20.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

magicrub, 2016-06-22:
Don't make sitl, run sim_vehicle.sh instead.

WickedShell, 2016-06-24:
sim_vehicle.sh won't fix this problem, it's compiler related. sim_vehicle.py might avoid it, as it leans on the waf build system rather then make to do the compilation. However if that fails there is a solution for make here: https://github.com/ArduPilot/ardupilot/issues/3586 see the comment from @galoulebougalou on Apr 27th.

closed 2016-06-24 by WickedShell

OXINARF, 2016-06-24:
It should be solved in master without any change needed, even for make

RoverHunter, 2016-06-24:
I am still having trouble compiling SITL on cygwin even with the  [#3586](https://github.com/ArduPilot/ardupilot/issues/3586) see the comment from @galoulebougalou solution.

I now get the following error:
/cygdrive/c/ardupilot/arduplane
$ make sitl
Checking modules
// BUILDROOT=/tmp/arduplane.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="a4277680" -I/cygdrive/c/ardupilot/libraries/AP_Common/missing -DMAVLINK_PROTOCOL_VERSION=2 -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE -DHAVE_CMATH_ISINF -DNEED_CMATH_ISINF_STD_NAMESPACE -DHAVE_CMATH_ISNAN -DNEED_CMATH_ISNAN_STD_NAMESPACE
%% Attitude.o
<command-line>:0:21: warning: "APM_BUILD_arduplane" is not defined [-Wundef]
/cygdrive/c/ardupilot/libraries/AP_Vehicle/AP_Vehicle_Type.h:36:41: note: in expansion of macro 'APM_BUILD_DIRECTORY'
 #define APM_BUILD_TYPE(type) ((type) == APM_BUILD_DIRECTORY)
                                         ^
/cygdrive/c/ardupilot/libraries/AP_BattMonitor/AP_BattMonitor.h:109:5: note: in expansion of macro 'APM_BUILD_TYPE'
 #if APM_BUILD_TYPE(APM_BUILD_ArduPlane)
     ^
<command-line>:0:21: warning: "APM_BUILD_arduplane" is not defined [-Wundef]
/cygdrive/c/ardupilot/libraries/AP_Vehicle/AP_Vehicle_Type.h:36:41: note: in expansion of macro 'APM_BUILD_DIRECTORY'
 #define APM_BUILD_TYPE(type) ((type) == APM_BUILD_DIRECTORY)
                                         ^
/cygdrive/c/ardupilot/libraries/AP_BattMonitor/AP_BattMonitor.h:130:5: note: in expansion of macro 'APM_BUILD_TYPE'
 #if APM_BUILD_TYPE(APM_BUILD_ArduPlane)
     ^
cc1plus: warning: unrecognized command line option "-Wno-unknown-warning-option"
Attitude.cpp: In member function 'void Plane::set_servos()':
Attitude.cpp:955:21: error: 'class AP_BattMonitor' has no member named 'overpower_detected'
         if (battery.overpower_detected()) {
                     ^
compilation terminated due to -Wfatal-errors.
C:/ardupilot/mk/build_rules.mk:11: recipe for target '/tmp/arduplane.build/Attitude.o' failed
make: *** [/tmp/arduplane.build/Attitude.o] Error 1

Is this normal with an official release? Is there any workaroud this problem?

OXINARF, 2016-06-24:
@RoverHunter from git hash in that log that's not any official release. Are you sure your local copy is properly updated? Having done a `make clean`?

RoverHunter, 2016-06-24:
I am trying to make sitl for release version 3.6.0 but I get the error I posted on the first message. Then, as recommended by WickedShell I have replaced on board_native.mk file:
EXTRAFLAGS += -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE
with:
EXTRAFLAGS += -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE -DHAVE_CMATH_ISINF -DNEED_CMATH_ISINF_STD_NAMESPACE -DHAVE_CMATH_ISNAN -DNEED_CMATH_ISNAN_STD_NAMESPACE

With this I now get the error I have shared on my second message on this post.

This is the only change I have made on the oficial 3.6.0 release to try to make sitl.

Any idea?

OXINARF, 2016-06-24:
The Plane 3.6.0 was at commit f8e0e91, from your log you aren't in that commit, you are in a very recent one. My guess is that you don't have a clean environment. You probably should try to clone the repository again and start over.

RoverHunter, 2016-06-24:
I have done just that with no sucess.
Git cloned the project, hard reset to 3.6.0 release. Open Cygwin, navigate to arduplane folder, make sitl. Waited for modules to download, make sitl again and the same message appeared:

 /cygdrive/c/ardupilot/arduplane
$ make sitl
Checking modules
// BUILDROOT=/tmp/arduplane.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="f8e0e91c" -I/cygdrive/c/ardupilot/libraries/AP_Common/missing -DMAVLINK_PROTOCOL_VERSION=2 -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE
%% tuning.o
<command-line>:0:21: warning: "APM_BUILD_arduplane" is not defined [-Wundef]
/cygdrive/c/ardupilot/libraries/AP_Vehicle/AP_Vehicle_Type.h:36:41: note: in expansion of macro 'APM_BUILD_DIRECTORY'
 #define APM_BUILD_TYPE(type) ((type) == APM_BUILD_DIRECTORY)
                                         ^
/cygdrive/c/ardupilot/libraries/AP_BattMonitor/AP_BattMonitor.h:109:5: note: in expansion of macro 'APM_BUILD_TYPE'
 #if APM_BUILD_TYPE(APM_BUILD_ArduPlane)
     ^
<command-line>:0:21: warning: "APM_BUILD_arduplane" is not defined [-Wundef]
/cygdrive/c/ardupilot/libraries/AP_Vehicle/AP_Vehicle_Type.h:36:41: note: in expansion of macro 'APM_BUILD_DIRECTORY'
 #define APM_BUILD_TYPE(type) ((type) == APM_BUILD_DIRECTORY)
                                         ^
/cygdrive/c/ardupilot/libraries/AP_BattMonitor/AP_BattMonitor.h:130:5: note: in expansion of macro 'APM_BUILD_TYPE'
 #if APM_BUILD_TYPE(APM_BUILD_ArduPlane)
     ^
cc1plus: warning: unrecognized command line option "-Wno-unknown-warning-option"
In file included from /cygdrive/c/ardupilot/libraries/AP_Math/AP_Math.h:16:0,
                 from /cygdrive/c/ardupilot/libraries/AP_GPS/AP_GPS.h:22,
                 from Plane.h:38,
                 from tuning.cpp:3:
/cygdrive/c/ardupilot/libraries/AP_Math/quaternion.h: In member function 'bool Quaternion::is_nan() const':
/cygdrive/c/ardupilot/libraries/AP_Math/quaternion.h:56:24: error: 'isnan' was not declared in this scope
         return isnan(q1) || isnan(q2) || isnan(q3) || isnan(q4);
                        ^
compilation terminated due to -Wfatal-errors.
C:/ardupilot/mk/build_rules.mk:11: recipe for target '/tmp/arduplane.build/tuning.o' failed
make: *** [/tmp/arduplane.build/tuning.o] Error 1

What could it be?

OXINARF, 2016-06-24:
How are you in a directory called arduplane? The directory should be ArduPlane

RoverHunter, 2016-06-24:
I was in a directory arduplane but now that I have tryed with ArduPlane, I get the same error message. Apparently, in Cygwin I can get inside both of them, despite the fact of having only ArduPlane folder.
It should be the same for cygwin, so I assume this is not the problem.

OXINARF, 2016-06-24:
That's because Windows doesn't differentiate between the two. But our code does care about that.

You don't get the APM_BUILD_TYPE warnings anymore right? Since you are not on master you now need to do the workaround in #3586

magicrub, 2016-06-24:
btw, using "make sitl -j4" will make this much less painful to test

RoverHunter, 2016-06-24:
I was finally able to compile sitl.
I believe the problem was name of the folder ArduPlane and not arduplane. Plus, I had to replace the board_native.mk EXTRAFLAGS as suggested.
Thank you all.

WickedShell, 2016-06-24:
Ah yes cygwin is in a weird spot where windows doesn't care about capitalization but its tools do... always cd to ardupilot/ArduPlane with the correct capitalization :)