ArduPilot/ardupilot #4375: handling terrain mission items when AP_TERRAIN_AVAILABLE=0

magicrub, 2016-06-24:
If you disable building AP_TERRAIN_AVAILABLE by setting 0 then it compiles fine but fails the SITL autotest run. Failure is from setting the waypoints as terrain points and the switch case to handle MAV_FRAME_GLOBAL_TERRAIN_ALT is gone so it returns MAV_MISSION_UNSUPPORTED_FRAME. The script sees the failure and retires a few times then times out and fails.

Here's an example of a failed build:
https://s3.amazonaws.com/archive.travis-ci.org/jobs/140077200/log.txt

possible solutions:
- change SITL autorun script to detect mission_ack results of MAV_MISSION_UNSUPPORTED_FRAME and if it was sending MAV_FRAME_GLOBAL_TERRAIN_ALT(2) then it should change the mission on-the-fly to another type such as relative height. (recommended)
- change Terrain mission item handler to treat as another type (not ideal)

OXINARF, 2016-06-24:
> change SITL autorun script to detect mission_ack results of MAV_MISSION_UNSUPPORTED_FRAME and if it was sending MAV_FRAME_GLOBAL_TERRAIN_ALT(2) then it should change the mission on-the-fly to another type such as relative height. (recommended)

I strongly disagree with this. Tests should be deterministic: our tests are too generic, but having a waypoint with terrain alt tests that the feature exists and is working well. If you change it to not care if the feature is there or not, then you are not testing the presence of the feature. Let's suppose that a bug would exist that, even with support for terrain compiled in it, it answered with unsupported frame. The test would not catch this.

In a technical point of view I also don't see it as easy to do: we just ask mavproxy to load the mission and I don't think mavproxy looks at the ACK message - so mavproxy would also need to be changed.

At best the test could be conditional based on the compilation - that way the test knows what to features are there or not, so it knows what to test or not. And in this specific case it would load a mission without the terrain alt waypoints.

WickedShell, 2016-06-24:
I'd strongly agree with Francisco here. And if terrain support isn't
compiled in then MAV_MISSION_UNSUPPORTED_FRAME is absolutely the correct
response.

magicrub, 2016-06-24:
Those are all valid points. Disregard my suggestions.

Maybe terrain support should be in the compatibility bits? Even if it is
there then autorun would have to detect it and load a different mission.

https://github.com/

OXINARF, 2016-06-25:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Terrain/AP_Terrain.cpp#L332

It's already there, although it doesn't seem correct to me that the capabilities bit is determined by a parameter. Also, Copter is always setting it, even when Terrain isn't compiled in - which obviously needs to be fixed.

Anyway, I would still prefer not to rely on the capabilities information (my opinion is that tests shouldn't be learning info in real time, it should all be known at compile time), but, as a workaround for now, I wouldn't oppose that.

rmackay9, 2016-06-27:
@OXINARF, thanks for the report on the copter capabilities bit!  I've fixed this in master now.
https://github.com/ArduPilot/ardupilot/commit/7edb6e76f0bc8470d47d05c3cd7ff4861a936ef7

OXINARF, 2017-06-04:
I'm closing this because:

- from the discussion, it is clear this is something for autotest and not a problem in the flight software
- there are several other features that can be disabled and aren't handled by autotest too, but listing all of them doesn't look productive to me
- we know our autotest suite is limited and there can be many improvements, but keeping an issue open for it won't bring any
- our current tests aren't broken, just someone trying to disable features and run autotest will have tests that depend on that feature fail
- like I said above, I think that tests should be deterministic and not rely on any answer provided by the flight software, the answer itself should be tested

If you think differently feel free to comment or re-open.

closed 2017-06-04 by OXINARF