ArduPilot/ardupilot #4381: Plane Bug: Zubax compass calibration and geotagging bug

jcooper21, 2016-06-26:
With Zubax GNSS 2 there is a GPS time stamp issue and compass calibration issue resulting in many compass errors and unable to geotag photos.

WickedShell, 2016-06-26:
There is a pr pending that should address this.

https://github.com/ArduPilot/ardupilot/pull/4300

hsteinhaus, 2016-07-12:
I guess it will - geotagging was the way I got aware about this bug. 

Potential workaround until the fix got merged: downgrade the compiler (however I cannot recommend a specific version).

WickedShell, 2016-09-15:
@jcooper21 was this fixed on plane 3.7.0 for you?

WickedShell, 2017-11-04:
Closing as #4300 should have fixed the handling of the time stamp overflows, and it's never been reported back as a problem since then. (And I've at least had a Zubax connected on the bench without seeing any problems).

closed 2017-11-04 by WickedShell