ArduPilot/ardupilot #4386: Unlimited Loiter in AUTO mission no longer works

rrr6399, 2016-06-27:
#### Issue details
When I specify the last step of an AUTO mission to be UNLIMITED LOITER, the copter lands at the end of the mission rather than loitering. If I simply revert to April 4th, 2016 version of 3.4, the AUTO mission works as expected.

#### Version
June 27th, 2016 fails
April 4th, 2016 succeeds
#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Hexacopter
#### Hardware type
Pixhawk

#### Logs

OXINARF, 2016-06-27:
Can you please post a log?

rmackay9, 2016-06-27:
@rrr6399, Thanks for all your testing and reports.

rmackay9, 2016-06-28:
I attempted to recreate this issue in about 5 different ways in the simulator but could not with master nor Copter-3.4-rc1.  This doesn't necessary mean there's no bug, just that I was unable to reproduce it (a log file would help with this).
Here's the mission I used: https://dl.dropboxusercontent.com/u/3852647/Copter-3.4-betatesting/copter_mission_test.txt

As a side note, there's really no need to include a Loiter-unlimited at the end of a mission because after the mission completes it will attempt to just Loiter at it's last location.  It will only land if it feels it has a "bad position" (i.e. GPS has gone bad which should triggera failsafe in any case). 

I'll close this issue because really support should be done in the discourse servers.  Our github issues list is really meant for confirmed bugs.   If we allow support issues to be added to the issue list, it'll simply become another support forum and the devs (including me) will lose track of confirmed bugs and enhancements.

closed 2016-06-28 by rmackay9

rrr6399, 2016-06-28:
Its very repeatable for me on Master. I'll find the log for this case.Here's the mission:

QGC WPL 110
0	1	0	16	0	0	0	0	35.639309	-120.646950	2476.667998	1
1	0	0	22	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1.000000	1
2	0	0	16	1.000000	0.000000	0.000000	0.000000	35.639297	-120.646942	3.000000	1
3	0	0	115	0.000000	60.000000	1.000000	0.000000	0.000000	0.000000	0.000000	1
4	0	0	16	1.000000	0.000000	0.000000	0.000000	35.639297	-120.646942	3.000000	1
5	0	0	17	0.000000	0.000000	1.000000	0.000000	35.639297	-120.646942	3.000000	1

I likely had my R/C receiver turned off as well..

The thing I don't like about the forum is that the issue will get lost and not addressed. Not too many are reading the forums. I guess I'll post the issue first and if it isn't addressed or it looks like it is a bug based on the discussion, I'll post the bug. Is that the desired process?

rrr6399, 2016-06-28:
This isn't a support issue as far as I can tell. The version in master works, but the version from April works fine with the exact same mission.

rmackay9, 2016-06-28:
@rrr6399, can you save the .txt file somewhere?  When i simply try to copy-paste the mission file above I get a floating point exception with mission planner.

Getting support is tough because it's nearly all done on a volunteer basis.  http://www.ardupilotinitiative.com provides paid support for companies but I'm definitely not trying to push that particular alternative especially for individuals.  For individuals I'm afraid I just recommend patience.  I certainly look through all the reported issues in the discourse forum before we go live.  Posting support issues here is really not appreciated actually.

rrr6399, 2016-06-28:
Why do you think its a support issue when the problem only shows up in the recent builds? I'm not changing anything on the GCS controller side. I posted the .bin file above. I'll post the .txt file as well.

Also, the way I'm getting into AUTO is this:
ARM, Guided Takeoff, AUTO.

I also noticed that the altitude goes way negative during takeoff probably due to ground effect. Not sure if that is impacting the current flight controller.

[image]

rrr6399, 2016-06-28:
Here's the .txt file dumped by MP:
[waypoints.txt](https://github.com/ArduPilot/ardupilot/files/336365/waypoints.txt)

rmackay9, 2016-06-28:
Ok, I think I see the issue.  The altitude type (3rd column of the mission file) is "0" which means absolute altitude (search for MAV_FRAME on this page: http://mavlink.org/messages/common).

In Copter-3.4 we support absolute, relative and terrain altitudes so you likely saw this problem appear when we added this support.  Here's a modified file with the altitudes all set to relative.
https://dl.dropboxusercontent.com/u/3852647/Copter-3.4-betatesting/rob-waypoints-rel.txt

rrr6399, 2016-06-28:
Thanks Randy. Does the http://www.ardupilotinitiative.com  take PayPal donations?

rmackay9, 2016-06-28:
@rrr6399, yes ArduPilot initiative accepts payments to info@ardupilotinitiative.com.

The ArduPilot non-profit can also accept payments now from [here](https://www.paypal.com/cgi-bin/webscr?cmd=_s-xclick&hosted_button_id=BBF28AFAD58B2).

Either is a-ok and no pressure of course but greatly appreciated if you do donate!

rrr6399, 2016-06-28:
All of the waypoints in the mission should've had their MAV_FRAME set to MAV_FRAME.MAV_FRAME_GLOBAL_RELATIVE_ALT, but the LOITER_UNLIMITED did not. Not sure why all of the reference frames came back as 0 from the Ardupilot or maybe MissionPlanner didn't set them? Anyway, I'll be testing this change today on RC1 and dev.

magicrub, 2016-06-28:
Randy. At my request Michael O has recently added support in MP for the
export kml button to also dump out the mission. There's an issue to also
dump out the params.

rrr6399, 2016-06-28:
Tested RC1 today with the correct ref frame specified for unlimited loiter and it worked fine. Thanks!