ArduPilot/ardupilot #4387: Land causes copter to fall out of sky when R/C is turned off

rrr6399, 2016-06-27:
#### Issue details
During testing recently, we've noticed that switching to Land mode at the end of a AUTO mission or even in guided mode with the R/C turned off will cause the copter to fall out of the sky many times and depending on the copter it may even be pitch and roll hard if only the throttle is specified with the R/C override message.

Even though this bug seems to be related to the ongoing bugs caused by the firmware not specifying nominal values for the yaw, pitch and roll when the R/C is turned off and the GCS has not overridden the R/C controls, I would expect LAND to not be dependent on the R/C settings.

#### Version
3.4

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Hex and Quad (Iris)

#### Hardware type
Pixhawk

#### Logs
TBD

OXINARF, 2016-06-27:
Log please

rrr6399, 2016-06-27:
Log uploaded.

lvale, 2016-06-28:
I believe this is a support question better addressed at http://discuss.ardupilot.org

It really depends on how your RC receiver reacts to the RC failsafe event. See here: http://ardupilot.org/copter/docs/radio-failsafe.html#receiver-and-flight-controller-set-up

OXINARF, 2016-06-28:
@lvale I understand your point, but I think that in no way we should crashing **if** we know we are in RC failsafe. I haven't looked at the logs, but that is exactly the reason why I asked for them.

lvale, 2016-06-28:
The logs show that only batt failsafe has been set, and to a very low value of 10.5, which considering the Voltage line I think we are dealing with a 4S battery, so 10.5 would give 2.6V per cell which is dangerous to say the least.

rrr6399, 2016-06-28:
I need to run missiobs solely with the GCS without the RC turned on. There is a bug related to having the RC turned off and issuing the LAND command. As you see from the logs, it almost behaves as a disarm when I issue LAND.

rrr6399, 2016-06-28:
The Battery FS was disabled on purpose as well.

lvale, 2016-06-28:
Batt Failsafe was ON.

Also your log terminates exactly when the LAND command is given.

[image]
[image]

rmackay9, 2016-06-28:
It's disarming the vehicle as soon as it's switched into Land mode.  I don't think it's related to a failsafe.  Definitely looks like a bug that we should fix asap.

OXINARF, 2016-06-28:
@rmackay9 Where are you seeing that?

[image]

What I see is that he changed to Land and it landed, after that, it disarmed.

OXINARF, 2016-06-28:
Ok, got it, you are talking about the last two lines. I would say that without LOG_DISARMED turned on I wouldn't jump to any conclusions here.

rmackay9, 2016-06-28:
@OXINARF, check out the vehicle's CTUN.Alt though.  it's still 7m off the ground.

OXINARF, 2016-06-28:
@rmackay9 I'm almost sure it didn't detect that it was off the ground again. We have an event for NOT_LANDED at the beginning but not another after the first land.

OXINARF, 2016-06-28:
Besides the event, I've checked the ap variable state and it indeed though it was landed. I still couldn't figure out why that was - throttle went above non-takeoff throttle which should have set land_complete to false.

peterbarker, 2016-06-28:
On Mon, 27 Jun 2016, Francisco Ferreira wrote:

> Besides the event, I've checked the ap variable state and it indeed though
> it was landed. I still couldn't figure out why that was - throttle went
> above non-takeoff throttle which should have set land_complete to false.

Isn't that assuming the land detector is running?

We land once, setting land_complete to true.

We never reset it, even when flying - I think that should change :-)

I think Copter::exit_mission is the problem; it just looks at 
land_complete before disarming the motors.

OXINARF, 2016-06-28:
> Isn't that assuming the land detector is running?

Yes, but it runs on the fast loop and I'm pretty certain that it hasn't stopped.

> We land once, setting land_complete to true.
We never reset it, even when flying - I think that should change :-)

I don't understand what you mean. We turn it to false when we think we are flying - but that check may need some improvement.

> I think Copter::exit_mission is the problem; it just looks at
land_complete before disarming the motors.

If you are landed and are exiting a mission why wouldn't you want to disarm the motors? It seems like the correct think to do.

rmackay9, 2016-06-28:
@rrr6399,
One contributing factor is that the vehicle is very overpowered.  So it looks to be getting off the ground at about 20% throttle.  The THR_MID value is set at 500 meaning that the controllers and land-detector code are expecting that it certainly won't takeoff at less than 25% throttle.
For now, could you please reduce the THR_MID parameter value to 300?

peterbarker, 2016-06-28:
On Mon, 27 Jun 2016, Francisco Ferreira wrote:

>       We land once, setting land_complete to true.
>       We never reset it, even when flying - I think that should change
>       :-)
> 
> I don't understand what you mean. We turn it to false when we think we are
> flying - but that check may need some improvement.

Yes, I think Randy's right; the output throttle wasn't exceeding the 
relevant threshold to indicate the vehicle has taken off.

>       I think Copter::exit_mission is the problem; it just looks at
>       land_complete before disarming the motors.
> If you are landed and are exiting a mission why wouldn't you want to disarm
> the motors? It seems like the correct think to do.

Sorry, we're probably not even crossing that code.

I actually misunderstood the way land_complete was updated and used.

OXINARF, 2016-06-28:
@rrr6399 Why are you using a dev version from April? You actually encountered a bug that was already fixed in 3.4-rc1...

The calculation I was doing for the non-takeoff throttle was based in the rc1 but he was using a dev version from April that indeed had a bug - it didn't account for THR_MIN.

@rmackay9 I leave it up to you, but I consider this solved: a mix of using a dev version with a bug and not setting THR_MID to a good value.

rrr6399, 2016-06-28:
Yes. I was testing the 3.4 version from April 4th since the dev version had a bug in the UNLIMITED_LOITER function of the AUTO mission where it landed rather than loitered.

Is THR_MIN the only info that is looked at to determine if the copter is flying and used to reset the land_complete variable. That might explain why the Iris worked fine as long as the R/C was turned on. 

If we had the R/C turned off (using the April 4th version again), the copter would either drop out of the sky or go to a max pitch and roll when landing was initiated. I'll have to use R/C override on all channels to prevent that.

rrr6399, 2016-06-28:
Peter's explanation perfectly explains why landing always worked from guided or auto the first time, but not the second time. Has the variable reset problem been fixed in dev? Could it be reset during arming?

OXINARF, 2016-06-28:
@rrr6399 First of all, you should read every post before answering, Peter already said he misunderstood the way land_complete was updated. So, no, Peter's explanation doesn't explain it and there was never a problem with land_complete.

Second, you are mixing two problems:

- the last issue in the flight where changing to Land caused it to disarm: it happened because the version you were using had a bug **and** because you had THR_MID set incorrectly
- the RC turned off: you have RC failsafe turned off! Which means that for ArduPilot as long as it receiving values from the receiver they are good values

rmackay9, 2016-06-28:
Ah, right, this isn't -rc1!  I totally didn't realise and had just assumed it was -rc1.

Rob, can you please upgrade to Copter-3.4-rc1 and reduce THR_MID to 300?  It really is best if you could stick with the official release candidates.  We can't really support versions taken straight from master at various periods of time.

I'll close this issue.

closed 2016-06-28 by rmackay9

rrr6399, 2016-06-28:
@rmackay9, I will try this off dev and RC1. I can't use them in general though due to the UNLIMITED_LOITER bug and that the Iris flew terribly with the version from the last few days. Are you making all buges fixed on the Copter-3.4 branch?

rrr6399, 2016-06-28:
@rmackay9 Is THR_MID the only variable you use to detect takeoff and that is used to reset the land_complete state? What happens when the copter has different weighted payloads on it? THR_MID will have to be adjusted for every new weight.

rrr6399, 2016-06-28:
Doesn't look like any work is being done in the Copter-3.4 branch. I'll try out RC1 and the current master tomorrow. I've broken a couple of legs and props during testing this so its a bit risky testing, but I'll try to keep the hover height low. 

BTW, it looks like the barometer isn't getting reset to 0 many times as well during arming.

rmackay9, 2016-06-28:
@rrr6399, all bug fixes are going into master and -rc2 will likely be rebased on master when it goes out.  I guess you see the issue here with using master but just to be clear, master changes every few hours and we can't make any guarantees that it's going to be flyable.  For example, I know there's an issue with Flip mode in master (it overshoots by 20degrees).  It's in master but we plan to correct it before -rc2 goes out.  If you just load up a version from master, it's even possible to catch a version that won't compile because you could, with a bit of bad luck, pick it up at the same time someone is doing a check-in of a long series of related commits.

I'll have a look at the unlimited-loiter issue.

The barometer isn't reset on arming.  In Copter-3.4 at least it sticks with the ekf origin which never moves.  The altitude above home should be reset on arming but there's a known issue with this not happening if the vehicle doesn't have GPS lock at the time of arming.

We should move these discussions to the discourse server.

rrr6399, 2016-06-28:
I'll try both. I had GPS lock and the altitude above home didn't get reset. Anyway, I'll try out RC1 and master tomorrow. (I pull the compiled versions off the firmware website so at least I'm good there.)

Why is THR_MID being used to reset the land_complete variable? Is that still the case?

rrr6399, 2016-06-28:
@OXINARF Would you please send me the commit date or github link to your checking when you fixed the bug you think is related to this? I can't find it in the git logs.

OXINARF, 2016-06-28:
@rrr6399 Just to make you happy: https://github.com/ArduPilot/ardupilot/commit/a5c55c3fbdf4f8135b744144768a2d658dd742ae

rrr6399, 2016-06-28:
Thanks @OXINARF ! I'll try it out RC1 and dev today to see if it works. I'm still concerned that a too high THR_MID setting can cause a disarm while landing.

OXINARF, 2016-06-28:
@rrr6399 Bad configurations can always cause problems... It's not the developer's fault if users don't configure their vehicles properly. We can make checks only to a certain point.
Also, there's a plan to calculate THR_MID (actually a new parameter, this one will disappear) automatically in rc2 - there's a bug in the current implementation of this so we're still not sure if it will make it or not.

rrr6399, 2016-06-28:
The issue is that it may change based in copter payload and it was never required before so many users may trip up on it. Can't the land_complete variable be reset upon disarming or arming?

I did test RC1 with my THR_MID set to 250 just to be sure and it worked as expected; thanks!

I looked at the dev version as well later, but was perplexed by the missing THR_MID variable so I aborted.

R-Lefebvre, 2016-06-29:
This situation should be considered for interaction with the Throttle/Collective function on TradHeli.