ArduPilot/ardupilot #4389: Support for N-Position switches via flexible channel ranges

lvale, 2016-06-28:
#### Issue details
Support for N-Position switches via flexible channel ranges. 

Currently the flight mode CH5 is the only that supports fixed ranges for multiple actions (different flight modes)

There is a limited support for 3 fixed positions switches on some AUX/Options channels, again with fixed range (max-mid-min)

So this could be split on 2 stages. 

1st stage - enabling flexible channel ranges
2nd stage - extend the flexible range options to the AUX/Options channels

#### Version
All

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
All
#### Hardware type
All

#### Logs
N/A

peterbarker, 2018-08-21:
Related issue: it would be nice to replace `FLAP_IN_CHANNEL` with an `RCn_OPTION` value - but `FLAP_IN_CHANNEL` is not a switch, it's a continuous input. 

It's also related to the elimination of RC_MAP.

peterbarker, 2018-08-21:
Sure.  How do we represent this information in the parameters nicely?
 
For the flight mode switch we have 6 magic parameters (at adjacent 
offsets, no less).  Adding N parameters to RC_CHANNEL... not really an    
option :-)

Some sort of indirection?  "virtual switches" as it were?

So

```
RC10_OPTION=4242  // RC_Channel::aux_func::virtualswitch1
RC_VS1_1=3  // RC_Channel::aux_func::SIMPLE_MODE  // RC VirtualSwitch1 position 1
RC_VS1_2=13 // RC_Channel::aux_func::SUPERSIMPLE_MODE  // RC VirtualSwitch1 position 2
RC_VS1_3=666 // RC_Channel::aux_func::INVOKE_ARMAGEDDON   // RC VirtualSwitch1 position 3
RC_VS1_4=4242 // RC_Channel::aux_func::BLOW_MIND   // RC VirtualSwitch1 position 4
.
.
.
```

You're still making a continuum discrete, of course.

Any other proposal?

peterbarker, 2018-08-21:
Ah.  Re-reading @WickedShell 's post - I think he's talking "in parallel" where @lvale is talking "in series".

"In parallel" is probably best done with RCn_OPTION2 or similar - at least I haven't come up with anything better.

lvale, 2018-08-21:
First thing would be to end the magic values and have a division in x parts of the configured range.
User would choose how many functions per channel up to 6? 8? 10? The intervals should be larger than the 2 x RCx_DZ. By default this would allow to maintain the current behaviour

Then for each configured RC channel, we would have a RC_OPTIONz where z would depend on the previously configured value above.

So, for each channel, by default we would have a new parameter RC_OPTIONS_PER_CHANNEL (that by default would be the same as we have now (6 for flight mode selection, 2 for all others)
So now remains how to set the available options on each channel without "exploding" the number of parameters. Considering that we could have xFF (256) possible options available the parameter to be added on each position I would say the easiest would be to store all on a single variable of length 2bytes per configured option x nr of options defined above.

peterbarker, 2018-08-21:
There are 2^16 options available, not 256.

I don't think we can multiply parameters like this - thus the indirection concept mentioned.

lvale, 2018-08-21:
Sorry, I wasn't clear
What I meant by available options are these, and with room to expand up to 256 
http://ardupilot.org/copter/docs/parameters.html#ch7-opt-channel-7-option

```

Values

0 | Do Nothing
2 | Flip
3 | Simple Mode
4 | RTL
5 | Save Trim
7 | Save WP
9 | Camera Trigger
10 | RangeFinder
11 | Fence
13 | Super Simple Mode
14 | Acro Trainer
15 | Sprayer
16 | Auto
17 | AutoTune
18 | Land
19 | Gripper
21 | Parachute Enable
22 | Parachute Release
23 | Parachute 3pos
24 | Auto Mission Reset
25 | AttCon Feed Forward
26 | AttCon Accel Limits
27 | Retract Mount
28 | Relay On/Off
34 | Relay2 On/Off
35 | Relay3 On/Off
36 | Relay4 On/Off
29 | Landing Gear
30 | Lost Copter Sound
31 | Motor Emergency Stop
32 | Motor Interlock
33 | Brake
37 | Throw
38 | ADSB-Avoidance
39 | PrecLoiter
40 | Object Avoidance
41 | ArmDisarm
42 | SmartRTL
43 | InvertedFlight
44 | Winch Enable
45 | WinchControl

```

peterbarker, 2018-08-27:
Are you suggesting we store `200.200.200.200.200.200.200.200.200.200` as a single parameter value somehow?  I'm not sure how that would work.  AFAICS that would require 10*8bits to store - and that's assuming an 8-bit option range which both tridge and I independently thought would be too limiting.  Additionally, apart from bitmasks, I can't think of other places we split parameters for different values.

lvale, 2018-08-27:
ok, so why 256 options are limiting ? I agree that the more the better, but considering the number we have today.....and unless we step up dramatically the number of options I don't see it as a problem in a couple of years. But if so then lets step it up to 65536 options (4bytes)?
Considering only a 2 byte for each option and 10 possible options on one channel that would be 20 bytes per rc_channel.

auturgy, 2019-01-27:
No reason to close. It’s a valid enhancement request.

tarantulito, 2020-07-30:
_No reason to close. It’s a valid enhancement request._

Having filled [#14941](https://github.com/ArduPilot/ardupilot/issues/14941) I fully agree. Simply opening the possibilities to 6, as for channel 5 (mode flight change), would permit (Ch7 RC7_OPTION as example):
-Assing Ch7 to a two position switch as now (RC7_OPTION=33 (brake)), repeating on three ranges.
-Assign Ch7 to a three position switch, repeating on two ranges:
  *Low: nothing (0).
  *Mid: brake (33).
  *High: motor emergency stop (31).
Of course, **in that order** for this example.
-Assign Ch7 to a mix of two two position switches (four functions), repeating, or a mix to a two and three position switches (six functions), done and operated very carefully.

To be modified:
-Code.
-Documentation.
-GCS's.

peterbarker, 2020-07-31:
On Thu, 30 Jul 2020, tarantulito wrote:

> Having filled #14941 I fully agree. Simply opening the possibilities to 6, as for channel 5 (mode flight change), would permit (Ch7 RC7_OPTION as
> example):

The flight mode channel is handled in a significantly different manner to 
channels on which we have other "auxillary functions" assigned.

Note that there's nothing magical about channel 5 - it's just the default 
on one of our vehicle types for changing modes.  You can use any channel 
as your flight-mode-channel.

We kind-of-sort-of have a table similar to the "virtual 6 position switch" 
I outline elsewhere in this issue - those are the FLTMODE1, FLTMODE2 etc 
fields - it's just the options are limited to changing modes rather than 
arbitrary channel options.

One problem with the concept of a "virtual switch" is how to handle 
three-position switches.  If it was just activating a boolean function - 
motor estop, for example - it would be relatively easy.  But there are 
plenty of auxillary options that use the mid-position to mean something 
special.

> -Assign Ch7 to a mix of two two position switches (four functions), repeating, or a mix to a two and three position switches (six functions), done and
> operated very carefully.

I do wonder at the need to convey so much information on a single RC 
channel - remembering that any option put onto such a channel has to be 
mutually exclusive from all the other options - you can't put landing gear 
and estop on the one RC channel as you do want to be able to operate them 
independently.

rmackay9, 2020-07-31:
I'm also not a huge fan of the idea of combining auxiliary switches together and/or having more positions on a single switch.  I think many new transmitters have huge numbers of switches and so I think it's a better idea for users who want a lot of features available to use one of these transmitters instead of trying to pack a ton of features onto a smaller number of switches.

I also think in general that the transmitter interface is quite limited and at some point to do really complicated things a ground stations (like the HereLink) is a better way.

tarantulito, 2020-07-31:
@peterbarker @rmackay9 : your comments and viewpoints are appreciated; read below.

@peterbarker

_You can use any channel as your flight-mode-channel_
For examples above FLTMODE_CH=5, which I suppose is universal.
BTW, not any: its possibilities are 5 to 8,
[image]
so if you use a three channels radio (rover) and want Ch3 for mode change you need a receiver with separate pins, attach a multiplexer and connect receiver output 3 to multiplexer input 5: having the possibility of using a PPM receiver, component added, cabling added and **less overall reliability**.

_...you can't put landing gear and estop on the one RC channel as you do want to be able to operate them independently._
With the implementation above (matrix, bit mapped modes) on a three position switch you can assign:
-Low: nothing.
-Mid: landing gear.
-High: landing gear **and** emergency stop.
Or with a mix on two two position switches (Gray coding):
00: nothing.
01: landing gear.
11: landing gear **and** emergency stop.
10: emergency stop.

@rmackay9 

_...many new transmitters have huge numbers of switches..._
But not channels: a 7 channels transmitter may have many switches, but only channel 7 (if 6 is analog) for RC7_OPTION and so only one possible function.
On the other hand, it should be documented that for using n options you need a 6+n channels radio, and a capable receiver.

_...to do really complicated things a ground station (like the HereLink) is a better way._
Even the ancient MultiWiiConf is better in these aspect: three pulse widths and simultaneous modes:
[image]

Cleanflight/Betaflight even better: permits modifying pulse width ranges:
[image]

About BEEPER in above figure:
-RCx_OPTION doesn't include continuous beeping. When loosing the copter I wonder if there is something more natural than continuously changing modes on the transmitter.
-On MP GCS, if assigning _Brake_ (33) to RC7_OPTION "_Mode changed to brake_" is heard when activated, but if assigning _Motor Emergency Stop_ (31) the motors stop but nothing is heard.
-Unless I miss something, there is no MAVLink message for the flight controller requesting a continuous beep on the GCS, such as on a dangerous mode as _Motor Emergency Stop_. So as an example if something here were implemented, the copter is far away and accidentally _Motor Emergency Stop_ is activated the GCS would remain silent.

@auturgy 

_No reason to close. It’s a valid enhancement request._

Replay: **I completely agree**.

peterbarker, 2020-07-31:
> _You can use any channel as your flight-mode-channel_
> For examples above FLTMODE_CH=5, which I suppose is universal.
> BTW, not any: its possibilities are 5 to 8,
> [image]

You can believe that if you like - but it's documentation, and is usually the case, wrong.  At least for Copter.

> so if you use a three channels radio (rover) and want Ch3 for mode change you need a receiver with separate pins, attach a multiplexer and connect receiver output 3 to multiplexer input 5: having [...]

Huh.  That sounds like fun.  I wouldn't recommend it, however.

> Or with a mix on two two position switches (Gray coding):

Problem with this is that mere mortals have to understand how it works.  And not just that, be able to debug it when it all goes wrong.

> _...many new transmitters have huge numbers of switches..._
> But not channels: a 7 channels transmitter may have many switches, but only channel 7 (if 6 is analog) for RC7_OPTION and so only one possible function.

I'm really not sure what you mean by "analog" here.  I guess you're referring to channel-6-tuning?

There is actually a way to accomplish what you're trying to do without vast number of parameters; a relatively simple LUA script would be able to do what you're describing with relative ease.

tarantulito, 2020-07-31:
@peterbarker
_You can believe that if you like..._
Long time ago, when beginning with a rover, I took that as absolutely true, so assembled the rover accordingly. I never tried FLTMODE_CH=3. I'll try: it implies a smaller PPM receiver, no multiplexer, less cabling and maintain reliability.

_I'm really not sure what you mean by "analog" here._
I refer to a Futaba 7C I use. It has a potentiometer for channel 6, which I use for gimbal pitch.

_...a relatively simple LUA script..._
I'd appreciate a brief orientation or URL on this. I looked at LUA scripts for WS2812's. As an example, two two position switches mix controlling Ch7 (four pulse widths) to get:
00: nothing.
01: brake.
10 and11: emergency stop.

tarantulito, 2020-08-03:
With a mix on two two position switches for Ch7, and a development version on a spare controller connected through USB:
ArduCopter V4.1.0-dev (30c8d7bf)
I tried:
_____________
function update()
  pwm7 = rc:get_pwm(7)
  gcs:send_text(0, "RCIN 7:" .. tostring(pwm7))
  if pwm7>500
  	if pwm7>1800 then
  		param:set('RC7_OPTION', 33)
  	else
  		if pwm7>1400 then
  			param:set('RC7_OPTION', 31)
  		else
  			if pwm7>1200 then
		  		param:set('RC7_OPTION', 0)
		  	else
		  		param:set('RC7_OPTION', 31)
		  	end
		end
  	end
  end
  return update, 500
end

return update()
_____________

(pwm7 = rc:get_pwm(7) seems to cause syntax error on beta and stable versions).

Pulling the USB cable and reconnecting, I find that RC7_OPTION starts with the chosen value on the radio switches, so this could be the way to go.

But, **does above script write every 500ms on the µP eeprom memory?**

peterbarker, 2020-08-05:
> But, **does above script write every 500ms on the µP eeprom memory?**

It won't cause writes - those values aren't written back to eeprom with `set` - you need `set_and_save` for that.

This isn't quite what I had in mind.  I *thought* someone had added a method to trigger auxiliary functions directly from LUA - sadly, I can't see any binding to do that :-(

tarantulito, 2020-08-05:
@peterbarker 
set ... set_and_save

Thanks. Experimenting on test bench with spare controller above script (with _set_) with the two two position switches mix for Ch7, transmitter on and off, on MP CONFIG-> Full Parameter Tree -> button "Refresh Params" -> pulling the USB cable (no supply to controller and receiver), etc, RC7_OPTION obeys to switches, but is not saved.

All that was for a development version (_ArduCopter V4.1.0-dev (30c8d7b)_). With the same script on latest beta:
_05/08/2020 11:51:46 : Frame: QUAD
05/08/2020 11:51:46 : RCOut: PWM:1-12
05/08/2020 11:51:46 : fmuv3 003C002D 50485011 20323450
05/08/2020 11:51:46 : ChibiOS: d4fce84e
05/08/2020 11:51:46 : ArduCopter V4.0.4-rc3 (d70e75b3)
05/08/2020 11:51:45 : **Lua: No scripts to run**_

FPVZaphod, 2025-01-06:
Digging in this old topic - it works with OSD Pages, why not with Flight Modes? You can freely choose, from which to whick uS a certain page is active.  Arducustom doubled the amount of possible modes to 11 or 12, this is already a great help. But the (IMHO) most elegant way is the one like Betaflight / iNav do.

hendjoshsr71, 2025-01-06:
This is a necessary feature for FPV or when you have limited switches like on my Radio Master Pocket.

It is one of the most annoying things when I switch a drone from BetaFlight to AP. Switching to a larger radio is unacceptable and in the end doesn't solve the problem as eventually you always run out of switches. 

**What Needs to Change?**
We would need to map RC Functions to RC Channels as is done in BetaFlight.

RC_Function 1 : would have three params for min_pwm, max_pwm, and RC_channel. 

I will note the capability of BetaFlight goes far beyond this and they also allow chaining functions together with logical AND, OR, & NOT.

FPVZaphod, 2025-01-07:
It became problematic when the vTols arrived. Fixed Wing Modes plus the hovering ones, and you are very tight with 6 modes. And then you burn one channel for each additional mode. It would be a great help already to just add one more parameter (amount of flight modes), and then spread them equally over the full range (1000-2000uS). Not even a need to define each range separately.

packetkiller, 2025-01-18:
This is a capability I think Ardupilot is one of the last flight controller software to address as it relates to handling of Channel PWM pulse widths. iNAV and BetaFlight have had this capability since this thread started. It should be a simple thing to have a two switch combo (3 way and 6 pos) provide 16 unique selections and not have to burn another channel. Using a bit mask and turning it into what ever is already in use should not be an issue. I program every day and this is one of the simplest of transitions. Don't let Ardupilot slip out of touch. Let's get this out of the 2015 era and keep Ardupilot relevant.