ArduPilot/ardupilot #4395: compiling for non-OS targets fails

night-ghost, 2016-06-29:
#### Issue details
because of line
`clock_gettime(CLOCK_REALTIME, &ts);`
in function uint64_t AP_HAL::Util::get_system_clock_ms() const
file libraries/AP_HAL/Util.cpp

which is call to OS function without any checks of HAL board

night-ghost, 2016-06-30:
moreover there is HAL function which returns system time in ms - HAL::millis(),  why anyone would need this new function in UTIL?

night-ghost, 2016-07-06:
changing function to 
```
uint64_t AP_HAL::Util::get_system_clock_ms() const
{

    /*struct timespec ts;
    clock_gettime(CLOCK_REALTIME, &ts);
    return ((long long)(ts.tv_sec * 1000 + ts.tv_nsec/1000000));
    */
    return AP_HAL::millis();
}
```
helps to solve this issue

lucasdemarchi, 2016-07-06:
millis() and get_system_clock_ms() are different. The first should return a monotonic time while the latter the realtime one.

This function was added by @rmackay9 and supposedly works on PX4. It works on Linux for sure.  There's a PR to change it from Util and group time functions together, but I guess you would need to provide an alternative implementation rather than just call millis(). What are you trying to do?

night-ghost, 2016-07-06:
> millis() and get_system_clock_ms() are different.
Yes but the difference becomes only if there is independent clock on the board.

>This function was added by @rmackay9
And definitely was not properly tested for all platforms.

>and supposedly works on PX4
Yes because PX4 uses Posix-compatible OS NuttX. But my [HAL for OpenPilot Revo](https://github.com/ArduPilot/ardupilot/pull/3874) don't! 
Linux is a posix-compatible OS too, sure.

If there no plans to drop HAL support and to become an ordinary Linux program *all* usage of board and OS functions MUST be done via HAL, and  function get_system_clock_ms too. For _my_ case replace it to millis() is an adequate replacement.

lucasdemarchi, 2016-07-06:
> And definitely was not properly tested for all platforms.

It was tested for all platforms we currently support in master

> Yes but the difference becomes only if there is independent clock on the board.
> Yes because PX4 uses Posix-compatible OS NuttX. But my HAL for OpenPilot Revo don't! 
> https://github.com/lucasdemarchi/ardupilot/commits/review-4350

ok. For those platforms in which it doesn't make sense, they can still override the function.

> If there no plans to drop HAL support and to become an ordinary Linux program all usage of board and OS functions MUST be done via HAL, and function get_system_clock_ms too.

get_system_clock_ms() is still part of HAL. Just because it is implemented on AP_HAL rather than AP_HAL_* doesn't mean you can't override it. All the currently supported boards have the same implementation, hence it was implemented just once.

> For my case replace it to millis() is an adequate replacement.

As I said, you can override it for your board. No need to break all the others.

lucasdemarchi, 2016-07-06:
ohh... and btw, there's a PR (#4350) changing exactly this function to support sitl on MacOS, although through and ifdef rather than overriding the implementation.

night-ghost, 2016-07-07:
>platforms we currently support in master

FlyMaple was supported in times of Copter 3.3 but now goes to AVR branch inspite of it uses almost the same STM32 and rather good for Rover and AntennaTracker...

>get_system_clock_ms() is still part of HAL. Just because it is implemented on AP_HAL rather than AP_HAL_* doesn't mean you can't override it.

I can not well know C ++, but to redefine the function it should be "weak", and to redefine the class method it should be "virtual". The AP_HAL::Util::get_system_clock_ms() is defined as `uint64_t get_system_clock_ms() const;`  so the compiler generates an error when I trying to override it.

Moreover when I change it to virtual I still got error
`AP_HAL/Util.cpp:56:38: error: 'clock_gettime' was not declared in this scope`

That's why it should be defined as `return AP_HAL::millis();` and redefined to use clock_gettime() in HAL of boards with OS.

night-ghost, 2016-07-14:
>It was tested for all platforms we currently support in master

commit 6e224158f1852b2c8f858567c00e081bec4bfde2  breaks compiling on all platforms for all tagets because contains few new constants, undeclared anywhere. So I see how your internal changes tested :)

OXINARF, 2016-07-14:
@night-ghost Really? I would swear that our platforms where compiling fine...

[image] [image]

 Those constants are very well declared, they come from MAVLink.

OXINARF, 2016-07-14:
Anyway, closing this as it is not a problem for any of the supported platforms - and Lucas already explained that it can be overridden.

closed 2016-07-14 by OXINARF

night-ghost, 2016-07-14:
> Lucas already explained that it can be overridden.

His method does not work in the real world!
 If I understand correctly that the closure of this issue means that support for HAL incompatible with POSIX completely abandoned? So PR with REVOMINI Hal now is garbage?

lucasdemarchi, 2016-07-14:
> If I understand correctly that the closure of this issue means that support for HAL incompatible with POSIX completely abandoned? So PR with REVOMINI Hal now is garbage?

It's one function that **future** non-posix ports have to ifdef out versus having to replicate the same function on all the other HAL implementations which right now are sitl, linux and px4.  It's just the "default" implementation that someone can perfectly ifdef out and implement its own on its hal implementation.

night-ghost, 2016-07-15:
>  have to ifdef out 

Ifdefs are evil! especially for C ++ where there is a more friendly mechanisms. Code with a lot of ifdefs is  awful and not readable. But if someone needs ifdefs it should be one define "BOARS_SUPPORTS_POSIX" or like

> having to replicate the same function on all the other HAL implementations

What a problem to include someting like "hal_posix.h" in all implementations?

> future non-posix ports

I have bad news for you: you are late, very almost. When OpenPilot died, people began outcome. CC3D board supports by CleanFlight so the question was not even, but the owners of the Revolution boards were out of work. If you would have included HAL for Revolution in copter 3.4 it would get a lot of new fans, but now there is a [fork](https://github.com/rs2k/raceflight) of CleanFlight that supports these boards, and of all the former OpenPilot users only newcomers can be interested in ArduCopter: "I'm lost how to get back? Wow, RTL button!".
But all gets even worse: already have a [fork](https://github.com/iNavFlight/inav) of Cleanflight which supports navigation, including a flight by the points and return to home. In a short time, it also begins to support the Revolution boards, and then it will be too late already definitively.

PS. Problem is that ALL code now written as only for POSIX:
```
from ./ardupilot-master.revomini2.build/libraries/AP_Module/AP_Module.cpp:22:
/usr/local/gcc-arm-none-eabi-5_2-2015q4/arm-none-eabi/include/sys/dirent.h:10:2: error: #error "<dirent.h> not supported"
 #error "<dirent.h> not supported"
```