ArduPilot/ardupilot #4397: Better RTL travel altitude

kudykam, 2016-06-29:
Tip fo better RTL travel altitude.
Current bahaviour of RTL is: if altitude lower then altitude for RTL, then ascend and return to home location, then descent vertically. If altitude is higher, then travel home at current altitude and descent.

Better option would be: if alt. higher, then travel home and gradually descent to RTL altitude (based on max descent speed), then over home location descent vertically. This will speed up time for returning to land and during forward movement, quad can descent faster then in zero speed (vortex ring). There is less chance then quad will drain battery before it lands.

OXINARF, 2016-06-29:
I believe that would a behavior too unpredictable. But it's up to @rmackay9 to decide if we should consider this as a future enhancement or not.

squilter, 2016-06-29:
I think it sounds reasonable. We are already making the assumption that everything above RTL_ALT is safe.

lvale, 2016-06-29:
I would say this assumes something that might be dangerous. If someone is flying higher than RTL_height, current behavior of returning at that height is correct.

OXINARF, 2016-06-30:
@squilter I would say that we assume that anything below current altitude or RTL_ALT isn't safe. Even at home you have the option not to land, precisely because in some circumstances we shouldn't assume that at home is safe to go down. I think we don't make any assumption about the state above RTL_ALT.

If this is proposed as different behavior **option** (turned off by default) then I think it would be fine. But OP proposes it as a replacement for current behavior.

kudykam, 2016-06-30:
RTL altitude should be always at safe height. I dont thing this approch would be more dangerous then current one. Anyway new option or replace default, it doesnt matter to me...

lvale, 2016-06-30:
15m is a good compromise as a default. Problem is that people tend to live with defaults.

R-Lefebvre, 2016-07-02:
I think this is a bad idea.  Certainly as the default or only behaviour, it would be bad.  And 15m is not a good compromise.  30m is minimum to clear trees of any size.  Especially if you are to leave some wiggle room to accomodate altitude loss due to flight speed affecting the baro.  And still not taking into account changing ground heights.

IMO, even offering this as an option is a bad idea.  If the extra few meters of descent over home is enough to cause you problems with your battery life, you've done something quite wrong already.

lvale, 2016-07-02:
@R-Lefebvre 15 or whatever height is something that we can easily adjust without disrupting the current behavior. On the proposed behavior I strongly agree with you. Very bad idea....

kudykam, 2016-07-03:
If you fly at 300m, when failsafe is engaged, then descending vertically those "few meters" will take you about 3,5 minutes (that can be saved) + time needed to travel distance over home location.

lvale, 2016-07-03:
You are unable to ensure that the oblique return path is safe. Use the appropriate values on the failsafes to ensure a safe return, or use Rally points.

rmackay9, 2016-07-04:
It's a good suggestion.  As to whether we make it the default or add an RTL_BEHAVE parameter is perhaps the only question.  Copter-3.4 also allows RTL-ing taking into account terrain altitude (enabled by setting TERRAIN_FOLLOW to "1") but it's behaviour is still not ideal.