ArduPilot/ardupilot #4407: make sure the AP_HAL_QURT is runs primarily on the DSPs?

Quintin-Z, 2016-06-30:
#### Issue details
_Please describe the problem, or desired feature_
i'm read the mk/borad_qurt.mk found the *.cpp just compile by arm cross tool
#### Version
_What version was the issue encountered with_
laster
#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2016-06-30:
Can you explain what the issue is? There's no explanation...

Also the issues are for confirmed bugs or future enhancements, not for support questions. For that you can use Gitter or post at the forum (http://discuss.ardupilot.org).

rmackay9, 2016-07-05:
The snapdragon build instructions are here on the wiki: http://ardupilot.org/dev/docs/building-for-qualcomm-snapdragon-flight-kit.html
No response so I'll close this.

closed 2016-07-05 by rmackay9