ArduPilot/ardupilot #4414: Quadplane: Q_Assist option to only help if airspeed too low.

iskess, 2016-07-02:
#### Issue details
Q_Assist is a great feature, but I am concerned about using up my very limited vtol endurance during standard autoflight. Since I only have about 2 mins of VTOL battery, I don't want the quad to inadvertently assist in climbing to waypoints for example. However, if the speed should ever drop below ARSPD_FBW_MIN, I do want the quad motors to kick in and save the day. 

#### Version
3.6

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover

#### Airframe type
3DR Aero Quadplane

#### Hardware type
Pixhawk

magicrub, 2016-07-02:
That's the current behavior. Vtol only kicks in when airspeed is low. Are
you seeing different behavior?

magicrub, 2016-07-02:
See q assist speed
http://ardupilot.org/plane/docs/parameters.html#q-assist-speed-quadplane-assistance-speed

iskess, 2016-07-02:
From reading that description and what was written about quadplane in the wiki, I was left with the understanding that there will be no power to the quad motors after the takeoff transition, unless Q_Assist is enabled. I haven't attempted a stall yet with Assist disabled, so perhaps I'm mistaken. I did test on the bench with the props off by switching to FBWA, the quad spins up, then I blow in the pitot tube until the quad shuts down.  When I stop blowing, the motors don't come back to like, it's just a conventional airplane at that point.

tridge, 2016-07-06:
If the airspeed stays above Q_ASSIST_SPEED then the VTOL motors will stay off for the whole fixed wing flight. So it should be set to a speed at which the fixed wing really will need assistance. If you never want the VTOL motors to assist in fixed wing flight then set Q_ASSIST_SPEED=0. 
If you set it to zero then it will still provide VTOL lift during transition, and will start the transition where it lowers the VTOL motors to zero over 5 seconds when you reach ARSPD_FBW_MIN.

iskess, 2016-07-06:
I had Q_Assist set to min speed. I was in FBWA at trim speed and then switched to auto. The next waypoint required a steep climb gradient and the quad motors fired up to assist. This is the behavior I would like the option of turning off.  Make sense?

iskess, 2016-07-09:
I wanted to add this quote from the wiki to explain the behavior that I think should be optional, while still preserving the Q_ASSIST at a speed just prior to stall. 

> "If you are flying in an autonomous or semi-autonomous mode then the quad motors will try to assist with whatever climb rate and turn rate the autonomous flight mode wants. In a manually navigated mode the quad will try to provide assistance that fits with the pilot inputs"

tridge, 2016-08-29:
I've clarified it in the docs.
Assistance will only be provided if the airspeed drops below Q_ASSIST_SPEED, or the attitude error goes above Q_ASSIST_ANGLE (once the PR for that goes in)
thanks for pointing out the ambiguity in the docs!

closed 2016-08-29 by tridge