ArduPilot/ardupilot #4424: Saving and retrieving Missions to/from the microSD card

lvale, 2016-07-04:
#### Issue details
Implementation of saving and retrieving an entire mission to and from the onboard sdcard.

Mission would be saved on SD card and possibly a Mavlink command to save and retrieve an entire Mission from the SD card.

As a cautionary step the first implementation would only allow this to happen when disarmed/on the ground. Missions before loading from the SD card would be required to respect some TBD parameters for validity

#### Version
New feature
#### Platform
[X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
All
#### Hardware type
All
#### Logs
N/A

freyapilot2002, 2016-09-01:
Is there any chance this is going to be implemented?
Everyone's comeback is always "use the distance trigger".
Distance trigger is a cool implementation but it is not always the answer.

freyapilot2002, 2016-09-01:
If it is not a global change is there a developer that is willing to change the code for a custom firmware?

magicrub, 2016-09-01:
@freyapilot2002 To get features added to Ardupilot, either in the master code base or not, there are a few options:

1. make an issue and hope someone volunteers their time to do it (like you've done here so far)
2. write it yourself
3. encourage another company to do it if they think it's worth their time (this is similar to # 1)
4. pay someone to write it - See http://ardupilot.org/dev/docs/common-commercial-support.html

Option # 4 is become more popular as there are more and more commercial demand for specific features that a wider audience does not benefit from but certain individuals/companies do

freyapilot2002, 2016-09-01:
Thank you for the great suggestions. I will pursue #4.

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2016.0.7752 / Virus Database: 4649/12920 - Release Date: 08/31/16

DonLakeFlyer, 2016-09-01:
Would be great to do this using standard mavlink messages (new ones) so it can become a common feature.

auturgy, 2019-01-27:
No reason to close. It’s a valid enhancement request.

reubenfinch, 2019-08-20:
Has there been any development on this? I'm unsure if the current storagemanager is even making full use of the larger RAM offerings in newer flight controllers. The Pixhawk4 has 512K, but we're only making use of around 128K from my reading of the code.

peterbarker, 2024-07-31:
We can do all of this using scripting; there are examples (or applets?) that do so.

We can also use SD card or storing these items just generally now.

closed 2024-07-31 by peterbarker