ArduPilot/ardupilot #4427: coverity: TECS flags

magicrub, 2016-07-05:
Here's an interesting TECS coverity warning:

> CID 91386 (#1 of 1): Identical code for different branches (IDENTICAL_BRANCHES)
> identical_branches: The same code is executed regardless of whether (*this)._flags.badDescent || (*this)._flags.underspeed is true, because the 'then' and 'else' branches are identical. Should one of the branches be modified, or the entire 'if' statement replaced?

Funny thing is it's flagging on [this](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.cpp#L411) but ignores [this](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.cpp#L398) which is identical.

OXINARF, 2016-07-05:
@magicrub It's not identical. What the warning is saying is that going through the _if_ path or the _else_ path, it does the same (so the if is not needed and could be removed). The other case you mention has no else, so the warning doesn't apply.

OXINARF, 2016-07-05:
And it has been there since the beginning: https://github.com/ArduPilot/ardupilot/commit/5b0129e02b4c816becae32d572ad5bf0030ff893#diff-b6ecb54e20c4b89219bde19a4b8e93eeR372

magicrub, 2016-07-05:
> And it has been there since the beginning: 5b0129e
> #diff-b6ecb54e20c4b89219bde19a4b8e93eeR372

oh wow, yeah you're right. That's so silly!

OXINARF, 2017-06-04:
Fixed with d8b58690ad56ca6cdbbdc075010a21d27e0fffe2

closed 2017-06-04 by OXINARF