ArduPilot/ardupilot #4429: AC_Avoid: stop-at-polygon fence is incorrect for acute angles

dricketts, 2016-07-05:
#### Issue details

If the polygon fence contains interior angles that are less than 90 degrees, it is possible to escape the fence. The bug is in AC_Avoid::adjust_velocity_poly(). This function iterates over all edges, repeatedly adjusting the velocity to be safe for the current edge. The problem occurs in the following scenario:

- The velocity is unsafe for the first edge in an acute angle, causing an adjustment.
- The adjusted velocity is unsafe for the second edge in the acute angle, causing a second adjustment.
- The adjustment after the second edge is unsafe for the first edge, but the first edge is not considered again.

rmackay9, 2016-08-29:
Bumping to AC3.5.  Maybe we can do a patch release (i.e. AC-3.4.1).

dricketts, 2016-08-29:
@rmackay9 Sorry this has taken me so long. It proved to be a bit trickier than expected, but I think I have a fix and hope to be done by next week. However, I think it's fine to bump this to 3.5 or 3.4.1.

rmackay9, 2016-08-29:
@dricketts, great, looking forward to it!

dricketts, 2016-09-11:
@rmackay9 I just submitted a patch for this: https://github.com/ArduPilot/ardupilot/pull/4813.

IamPete1, 2021-03-24:
Acute angles are still a issue for velocity avoidance, 
[image]

rmackay9, 2021-03-24:
My guess is we need to do something similar to what we've done for obstacle avoidance which is to capture the largest correction in four quadrants (+ve x and y, -ve x and y) and then add the +ve and -ve values for x and y to get a final x and y.  The issue may also happen across types (i.e. mixing proximity and fence) although I'm not sure.