ArduPilot/ardupilot #4435: RTL_Radius supercedes WP_LOIT_RAD ArduPlane 3.5.3

falconunmanned, 2016-07-06:
Arduplane 3.5.3

We were flying an unlimited loiter waypoint with an WP_LOIT_RAD of 400m.  We switched the RTL_RAD to 150 and the aircraft changed its radius to 150 while in the Auto mode for the loiter unlimited waypoint.  When the full parameter list was viewed the WP_LOIT_RAD was still 400m but doing a 150m radius circle.

#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2016-07-06:
confirmed, but I think it was by design due to a request at some point in the past. Does anyone remember? Loiter-unlimited was treated differently for some reason.

here's the logic: https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/commands_logic.cpp#L606
for NAV_LOITER_UNLIMITED the radius is never zero, its either the direction size or it's RTL_RADIUS. It never looks at WP_LOITER_RAD like _TIME and _TURNS does.

WickedShell, 2018-03-10:
Just came across this, I'd say that behaviour is *very* surprising and unintuitive. I'd support removing the special case code that uses RTL's radius here.

magicrub, 2018-03-10:
I think the old old old behavior must have been an RTL event stayed in AUTO and executed loiter unlimited over home. Would have been equivalent behavior. I remember seeing this back when I was making all the loiters share common code and had to treat this one a little different to retain the old behavior.

I vote to remove the special behavior.

Naterater, 2019-01-04:
Confirmed in SITL, but only for LOITER_UNLIMITED.

IamPete1, 2021-07-26:
fixed by https://github.com/ArduPilot/ardupilot/pull/14642

closed 2021-07-26 by IamPete1