ArduPilot/ardupilot #4436: AP 3.6.0: Crash detection triggered in flight after takeoff and disarmed flying plane

marcmerlin, 2016-07-06:
Ok, first I'll start with the good news: the resulting crashes were soft and did not cause damage (but I also didn't trust AP and used a smaller throw away plane, just to be sure :) ).

https://youtu.be/Y7B0IwQUItM shows the problem:
1) switch to auto
2) throw plane
3) auto takes over, starts the motor
4) just after that it detects a crash while the plane is in flight and disarms the motors
5) plane goes down, and thankfully lands ok

grep CRASH 23.txt
CRASH_ACC_THRESH 0.000000
CRASH_DETECT    1.000000

I did not touch the CRASH options. so those are default.
I don't see why crash detection would trigger if CRASH_ACC_THRESH is 0.
Obviously triggering in flight is bad, thankfully it happened in a way that it was not a big deal in my case.
[23_auto_crash_disarm.txt](https://github.com/ArduPilot/ardupilot/files/350982/23_auto_crash_disarm.txt)

Full log (auto is at the end of flight): http://marc.merlins.org/tmp/23_auto_crash_disarm.bin

marcmerlin, 2016-07-06:
I'm seeing this in the logs if that helps. Flight lasts 0.5sec: AUTO arms the plane, starts the motor, flies just a bit, and then crash is wrongfully detected while the plane is flying and motor gets disarmed.
@magicrub , since I see you touched that code :), could you have a look?

2016-07-06 09:45:40.54: MSG {TimeUS : 2707696488, Message : Triggered AUTO. GPS speed = 1.2}
2016-07-06 09:45:40.54: SONR {TimeUS : 2707696758, Dist : 0.0, Volt : 0.0, Cnt : 0, Corr : 0.0}
2016-07-06 09:45:40.54: RFND {TimeUS : 2707696767, Dist1 : 0.0, Dist2 : 0.0}
2016-07-06 09:45:40.54: ATT {TimeUS : 2707696788, DesRoll : -2.12, Roll : -0.89, DesPitch : 6.63, Pitch : 11.49, DesYaw : 0.0, Yaw : 359.49, ErrRP : 0.19, ErrYaw : 0.06}
(...)
2016-07-06 09:45:41.26: NKF1 {TimeUS : 2708417734, Roll : -0.87, Pitch : -1.67, Yaw : 4.44, VN : 5.89570474625, VE : -0.00292715989053, VD : 1.00922417641, dPD : -0.088161110878, PN : 10.8583459854, PE : -0.976049244404, PD : -0.904401123524, GX : -0.19, GY : -0.01, GZ : -0.28}
2016-07-06 09:45:41.26: NKF2 {TimeUS : 2708417734, AZbias : 13, GSX : 0.21, GSY : 0.12, GSZ : 0.47, VWN : -2.1, VWE : 0.66, MN : 244, ME : 61, MD : 454, MX : 119, MY : -6, MZ : -81, MI : 0}
2016-07-06 09:45:41.26: NKF3 {TimeUS : 2708417734, IVN : 1.05, IVE : 0.17, IVD : 0.05, IPN : 0.01, IPE : 0.11, IPD : 0.46, IMX : 2, IMY : 3, IMZ : 10, IYAW : -6.45, IVT : 0.0}
2016-07-06 09:45:41.26: NKF4 {TimeUS : 2708417734, SV : 0.2, SP : 0.0, SH : 0.03, SM : 0.07, SVT : 0.0, errRP : 0.00176875188481, OFN : 0, OFE : 0, FS : 0, TS : 16, SS : 9023, GPS : 0, PI : 1}
2016-07-06 09:45:41.26: NKF6 {TimeUS : 2708417734, Roll : -0.04, Pitch : -0.07, Yaw : 0.74, VN : 5.35655260086, VE : -0.0271600782871, VD : 1.06483888626, dPD : -0.142091035843, PN : 10.1366224289, PE : -0.896269857883, PD : -0.611718893051, GX : 1.85, GY : 0.56, GZ : 0.38}
2016-07-06 09:45:41.26: NKF7 {TimeUS : 2708417734, AZbias : -5, GSX : -0.58, GSY : 0.45, GSZ : 0.71, VWN : -2.33, VWE : 0.63, MN : 241, ME : 60, MD : 450, MX : 118, MY : -1, MZ : -75, MI : 0}
2016-07-06 09:45:41.26: NKF8 {TimeUS : 2708417734, IVN : 0.66, IVE : 0.11, IVD : 0.11, IPN : -0.18, IPE : 0.16, IPD : 0.67, IMX : 1, IMY : 2, IMZ : 11, IYAW : -10.84, IVT : 0.0}
2016-07-06 09:45:41.26: NKF9 {TimeUS : 2708417734, SV : 0.13, SP : 0.0, SH : 0.04, SM : 0.12, SVT : 0.0, errRP : 0.00157724122982, OFN : 0, OFE : 0, FS : 0, TS : 16, SS : 9023, GPS : 0, PI : 1}
2016-07-06 09:45:41.26: AHR2 {TimeUS : 2708417892, Roll : 0.48, Pitch : -1.84, Yaw : 8.27, Alt : -0.0599999986589, Lat : 37.411774, Lng : -121.9948366}
2016-07-06 09:45:41.26: POS {TimeUS : 2708417938, Lat : 37.4117835, Lng : -121.9948341, Alt : -1.09000003338, RelAlt : 0.611718893051}
2016-07-06 09:45:41.26: CTUN {TimeUS : 2708417952, NavRoll : -0.92, Roll : -0.04, NavPitch : 6.58, Pitch : -0.07, ThrOut : 98, RdrOut : -64, ThrDem : 98}
2016-07-06 09:45:41.26: NTUN {TimeUS : 2708417962, WpDist : 6.89313745499, TargBrg : -6.3, NavBrg : -6.3, AltErr : 49.38, XT : 0.0, XTi : -0.0141594354063, ArspdErr : 0.0}
2016-07-06 09:45:41.26: RCIN {TimeUS : 2708418019, C1 : 1492, C2 : 1503, C3 : 2016, C4 : 1501, C5 : 1494, C6 : 1091, C7 : 1868, C8 : 1552, C9 : 2016, C10 : 1501, C11 : 1948, C12 : 1419, C13 : 0, C14 : 0}
2016-07-06 09:45:41.26: RCOU {TimeUS : 2708418095, C1 : 1517, C2 : 1295, C3 : 1702, C4 : 1502, C5 : 1517, C6 : 1091, C7 : 1868, C8 : 1552, C9 : 2016, C10 : 1501, C11 : 1948, C12 : 1419, C13 : 1500, C14 : 1500}
2016-07-06 09:45:41.26: RSSI {TimeUS : 2708418113, RXRSSI : 0.948000013828}
2016-07-06 09:45:41.26: VIBE {TimeUS : 2708418129, VibeX : 1.43768131733, VibeY : 4.83131551743, VibeZ : 2.53910207748, Clip0 : 19, Clip1 : 31, Clip2 : 0}
2016-07-06 09:45:41.26: MSG {TimeUS : 2708418206, Message : Throttle disarmed}
2016-07-06 09:45:41.26: ARM {TimeUS : 2708418271, ArmState : 0, ArmChecks : 1}
2016-07-06 09:45:41.26: MSG {TimeUS : 2708418759, Message : Crash detected}

magicrub, 2016-07-06:
already looking at it. It's hard to follow because you have 24 mode changes. Sheesh! Also, are you sure this is the right log?
1) I see a good flight, no problems on takeoff
2) I don't see the crash flag going true - ever

marcmerlin, 2016-07-06:
Sorry it was at the end of other flights and I didn't reboot in the middle, but I already found you the right offset and pasted it, see my previous comment:
2016-07-06 09:45:41.26: MSG {TimeUS : 2708418206, Message : Throttle disarmed}
2016-07-06 09:45:41.26: ARM {TimeUS : 2708418271, ArmState : 0, ArmChecks : 1}
2016-07-06 09:45:41.26: MSG {TimeUS : 2708418759, Message : Crash detected}

magicrub, 2016-07-06:
So, I never see the crashed flag go true(1) because it disarms and you have the log bitmask set to disable logging when disarmed.

Well, this behavior is certainly not correct. The thing that puzzles me is I don't see STAT.isFlying going true(1) on your takeoff. That means the plane never thought it was flying, which is a requirement to crash. The crash detection logic is right here:
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/is_flying.cpp#L212

A couple things that should have prevented this is:
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/is_flying.cpp#L80
and
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/is_flying.cpp#L199

marcmerlin, 2016-07-06:
Thanks for having a look. Right, new airframe, didn't change the default log bitmask to log when disarmed. I'll change that.
1) since the crash is detected within less than 1sec after auto turns the motor on, is it possible that stat.is_flying does not have the time to get logged?
2) I see why it shouldn't have triggered according to the code, but on top of that, I have 
CRASH_ACC_THRESH 0.000000
CRASH_DETECT 1.000000
Since the threshold was 0, shouldn't that also have disabled the crash detection regardless of everything else too? (at least the documentation implies so). 
Obviously I'll turn CRASH_DETECT to 0 on my next test, but as is, it should already not be enabled, right?

marcmerlin, 2016-07-06:
Looking at https://github.com/ArduPilot/ardupilot/blob/a80a87eb09f4ac0186ba0d4d962111ddedd668fd/ArduPlane/is_flying.cpp#L59
crash_accel_threshold inhibits crash_state.impact_detected

but without reading in many details (maybe I missed something)
https://github.com/ArduPilot/ardupilot/blob/a80a87eb09f4ac0186ba0d4d962111ddedd668fd/ArduPlane/is_flying.cpp#L209
seems to imply that crashed can be set for a variety of reasons and apparently
https://github.com/ArduPilot/ardupilot/blob/a80a87eb09f4ac0186ba0d4d962111ddedd668fd/ArduPlane/is_flying.cpp#L291
can disarm motors regardless of the value of CRASH_ACC_THRESH.
Is that correct?

magicrub, 2016-07-07:
CRASH_ACC_THRESH see the [description](http://ardupilot.org/plane/docs/parameters.html#crash-acc-thresh-crash-deceleration-threshold).

X-Axis deceleration threshold to notify the crash detector that there was a possible impact which helps disarm the motor quickly after a crash

This just notifies the crash detector to skip through is_flying low-pass decay if we think we hit something. It won't trigger a crash. And again, if you're never flown then you should never have crashed. I'll investigate further.

marcmerlin, 2016-08-06:
Howdy. Just curious if you were able to make more sense out of this? Should I continue to fly with crash detection off for now, or is there a way not to trigger it when I'm not crashing?

magicrub, 2016-08-06:
Fly without for now, I haven't investigated at all yet. Thanks for the reminder.

tomarigr, 2016-08-24:
Hello, i stumbled uppon this issue, i dont know if this helps at all since i dont have logs nor version info but, i remember at about 20 nov 2015, having latest stable build of copter and mp, i was playing whith mp with just pixhawk connected. No motors no external gps nothing, so i wasn't even armed. When moving the pixhawk a certain way, mp was reporting a crashed alert and wasn't even recovering from this state unless u disconnect and reboot pixhawk. This never happened midflight though, but we can tell for sure that this version back around 20/11/2015 not only triggered crash while not flying, it triggered even disarmed. Maybe its a bug left from that days???
One week ago and while having 3.3.3 i tried to replicate it with the vehicle fully assembled and couldnt

mauro18421, 2017-01-20:
Hi.

I'm trying to set up a plane with a pixhawk and I'm having this issue. On an auto takeoff
 the plane armed and disarmed on les than a second.
It looks like I can disable the crash detection but ......

thanks

magicrub, 2017-01-20:
Not good. I haven't flown an aircraft for a couple of months, I'll need to get one and investigate this!

fredrikfalkman, 2017-10-10:
I seem to have a similar behavior (in 3.8.1) if I populate TKOFF_THR_MINACC with a value other then 0 and set CRASH_DETECT = 1.

In earlier versions, I used to set TKOFF_THR_MINACC = 5 to be able to start the motor with a mild shake and then throw the plane away when in Auto. 

Now immediately after the motor starts it detects a crash if CRASH_DETECT = 1, CRASH_ACC_THRESH = 25

If CRASH_DETECT = 0, then this shake to start works.

If TKOFF_THR_MINACC = 0 and CRASH_DETECT = 1, then I get motor start on Arm, but I can't initiate a crash however hard I shake, even if I set CRASH_ACC_THRESH very low, i.e. = 4.

magicrub, 2017-10-10:
@fredrikfalkman This is expected behavior. The TKOFF_THR_MINACC is intended to be used on bungee or catapult launches.
TKOFF_THR_MINACC > 0 means that after arming the throttle is inhibited until the specified forward x acceleration is detected.

CRASH_DETECT = 1 with TKOFF_THR_MINACC  > 0 means it is detecting failed bungee/catapult launches and cuts throttle very fast if it does not detect itself in the after after the acceleration. This is important because larger aircraft are the ones who use bungee/catapult and you don't want the motor/prop flopping around all crazy on the ground on a failed launch.

TKOFF_THR_MINACC = 0, CRASH_DETECT = 1 means a normal takeoff, and those takeoffs can't "crash" until you've flown.

What we need to add is TKOFF_THR_MINACC  > 0 with CRASH_DETECT=2 where it has a much longer delay (5s?) allowing you to shake, then hand-launch but still shutoff if you don't launch it in time

magicrub, 2017-10-10:
@fredrikfalkman  created https://github.com/ArduPilot/ardupilot/issues/7061 to track that new feature.

fredrikfalkman, 2017-10-10:
@magicrub Thanks, makes sense.