ArduPilot/ardupilot #4438: Altitude Use via LIDAR

Piesang86, 2016-07-07:
Hey guys,

I was wondering if there is someone that could help me out.
I've got a LIDAR in a PLANE/FIXED WING that I would like it to use it for altitude while flying.
I'm flying in areas where altitude varies between 20m-30m in altitude in a area of around 50ha. 
I need to fly at 70m in order to do plant counting, so I need to keep a relative altitude of around 70m. If the altitude gets too high (mission planner only lets you use a fixed altitude) that would mean I'll have to fly at 100m at some points. That would generate inconclusive data for the farmers for plant counting.

Is there someone that would help out to enable the PLANE to USE the LIDAR for ALTITUDE purposes?
Thanks in advance

rmackay9, 2016-07-07:
I'm not a plane developer but some thoughts based on my experience with Copter:

- I'd recommend trying the existing terrain following feature which relies on google earth to be sure that it's not good enough.  http://ardupilot.org/copter/docs/common-terrain-following.html
- it may be possible to improve the google-earth based terrain following by reducing TERRAIN_SPACING parameter from "100" (i.e. 100m) to "30".  This is untested but should improve the granularity of the terrain data sent from the ground station.
- google earth based terrain following is only available on the pixhawk.
- I suspect 70m is near the limits of the lightware SF11C lidar (120m) once you take into account the slightly optimistic range quoted and also remembering that we need extra room to compensate for the vehicle tilt (i.e. if tilted over at 45 degrees, a reading of 120m would mean the vehicle is actually flying at about 85m I think).

Piesang86, 2016-07-08:
Google earth has the same info that your GPS uses. That being said, it doesn't matter how you try to configure it with Google maps, it would be around 10m out on flat surfaces and 20m vertical (height) because of Google earth being a photo taken from space. We also do survey work, that's how we figured that out. Also it's not regularly updated, like a mine that blasts for example.

The plane using the LIDAR would be the most accurate measurement, keeping consistent altitude, even if it's out by 5m, that's way better than 20m+5m

gelazanskas, 2016-09-27:
Is anyone working on this? :)

magicrub, 2016-09-27:
Not that I know of. If this is important to you and/or your business then there are ways to expedite developers to allocate time to work on new features like this.

IronDome2, 2017-02-11:
Just thinking out loud, how would the LiDAR recognize that terrain is about to change ahead of it? Also if you are flying across sudden terrain change such as valleys and gullies, how would the plane react? I think the use of generic elevation data such as that provided by google is the way to go, because then perhaps, the flight controller can tell ahead o time that the terrain is about to start rising or going down, than than noticing the change while already right above the area. Also, i'm not a developer by all means, I just happen to think so as presented. i'm interested in testing the existing terrain following feature.

maaz1113, 2018-06-20:
Hello everyone. 
I am using a Lidar LW20 with Pixhawk cube on my Hexacopter for terrain following. It is working fine, but the problem is that the response is too slow. the copter hovers for a few seconds over the roof top before changing its altitude.I am experiencing a delay of about 2-3 seconds. Can someone help me in solving this problem?
-thanks

Naterater, 2019-01-05:
Although I have heard a lot of negative comments about this from the developers because we have a fear of collision avoidance in rising terrain, I still think this is a very valuable feature. If anything, a feature like this could be used to verify SRTM data. I personally would be using this in rolling hill terrain where abrupt climbs are not necessary, and it is good to always stay at the legal altitude limit above ground for high endurance missions. This feature could potentially correct for Baro drift when comparing SRTM data to the current altitude estimate from LiDAR.