ArduPilot/ardupilot #4441: How many timers are used to generate your PWM signals

bfishman, 2016-07-08:
Hey folks.  Awhile back I started building a simple arduino FC from scratch to help me understand the basics. An interesting problem I encountered was how to generate multiple PWM signals with the limited number of timers I had available on my Arduino Uno.  I came up with a solution that allowed me to generate theoretically any number of unique PWM outputs (limited by your number of IO pins!) using a single timer - in this case, the 16 bit timer for better precision.

I started to dig through the Arducopter code but I'm new to this repo and my head started swimming a bit. So I was wondering if someone could give me a short explanation of how this library generates PWM signals (but no need to stray away from technical details, I'm somewhat familiar with arduino programming and PWMs).

If there's a chance that my technique may be more efficient (either in terms of execution time and/or hardware utilization), then I would be more than happy to write up a detailed description of the technique and post my code for discussion.

No rush on this.. I hate to say it but I've been "sitting" on this code for almost 4 years now, always wanting to post it for discussion, but life and limited hobby-time focus moved on and I've only just recently come back to this wonderful project.

Cheers!

hiro2233, 2016-07-08:
Hi @bfishman for you know, Ardupilot development for 8bits hardware are no yet supported anymore. Right now, development are with 32Bits hardwares like DSP, SoC processors systems running under NuttX, Linux and others OS.  But if you want study AVR 8bits microcontrollers and how it's work you can see the OLD branch at this point: 

https://github.com/ArduPilot/ardupilot/tree/master-AVR

Begin estudiying the libraries/AP_HAL_AVR path folder libraries for AVR microcontrollers, it's optimized for atmega2560 microcontroller, for specific avr familiy you need to do somethings modification in the code.

bfishman, 2016-07-08:
Oh, thank you @hiro2233.  I did not realize development on 8 bit systems was no longer supported.  Maybe that explains why I didn't recognize any of the code :)

OXINARF, 2016-07-08:
The issues list is for bugs and feature requests. This is a (development) question and those should be posted at the forum (http://discuss.ardupilot.org) or in Gitter.

closed 2016-07-08 by OXINARF