ArduPilot/ardupilot #4442: Plane differential spoilers additional mixing options

raptortrias, 2016-07-08:
#### Issue details
Currently diferential spoilers with elevon_output move in both roll and pitch.

For high aspect ratio flying wings this can be a problem especially in pitch response as it effectively reduces the pitching moment (when full span elevon/spoilers are used).

 It would be nice to have additional control of how the spoilers move as percentage mixing in roll and diffferent for pitch

#### Version
3.6.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover

#### Airframe type
Flying wing

#### Hardware type
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

Naterater, 2019-01-04:
Is this still an issue with the new additions - crow flaps, etc?

IamPete1, 2020-07-28:
With the new [Differential Spoiler options](https://ardupilot.org/plane/docs/differential-spoilers.html#differential-spoiler-options) I think this is fixed

closed 2020-07-28 by IamPete1