ArduPilot/ardupilot #4448: Plane V3.6.0 reads Spektrum SRXL input incorrectly

jagrandle, 2016-07-11:
#### Issue details
I have been using a genuine 3DR Pixhawk and a Spektrum AR7700 receiver with no problems in my existing aircraft. The receiver is connect to the Pixhawk via its SRXL output.

I have been using Plane V3.5.2 with no problems.

I am now trying to replicate this setup for a second aircraft. Under Install Firmware the latest version of Plane is 3.6.0 Tried to find 3.5.2 under the Previous Versions but it was not visible. Next most recent version was 3.5.0

I Installed 3.6.0 but the throttle channel and Flight Mode Channel (5) kept bouncing around in terms of the measured values. I recalibrated Radio twice but pointless when the values keep bouncing around. I plugged a servo in to the Gear Channel (5) and there were no jumps in the output. I don't think there is a problem with the receiver.

I then tried an identical new receiver and had exactly the same problem. The servo plugged in directly was also fine,.

I then installed Plane version 3.5.0 and it works perfectly, no jumping on Throttle or Flight Mode.

I can only conclude that something to do with reading the Spektrum receiver port changed in version 3.6.0 even though there is no mention of it in the the revision notes.

I hope somebody can help. My preferred solutions in order would be..

1) Can somebody please tell me how I can see version 3.5.2 to install it.
or
2) Can somebody tell me how to get version 3.5.2 off my existing Pixhawk and copy it to my new one
or
3) Can somebody please fix version the code and release a new version.

Thanks,

#### Version
3.6.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover

#### Airframe type
Skywalker - Conventional lane

#### Hardware type
Pixhawk - Genuine 3DR

#### Logs
_Please provide a link to any relevant logs that show the issue_

jagrandle, 2016-07-23:
I have just upgraded Mission Planner to version 1.3.39 and a larger number of firmware versions are available under the "Load Previous Version" option than there was previously. It seems to check what versions are available, so I have no idea why this would change with the new version of Mission Planner, but it does !!

At least now I can load 3.5.2 Plane.

Hope you guys fins the bug in reading the Spektrum receiver in V 3.6.0

Cheers

WickedShell, 2016-10-17:
This is fixed in master and will be on the next release.

closed 2016-10-17 by WickedShell

magicrub, 2016-10-17:
fixed via: https://github.com/ArduPilot/PX4Firmware/pull/68

jagrandle, 2016-10-18:
Thanks so much guys. I really appreciate the help of all involved.
Jag

blin08, 2017-08-30:
I have just been trying to get an SRXL receiver to work with my pixhawk. Any firmware version that I load past Version 3.5.3 has the same radio input problem described above.

magicrub, 2017-08-31:
Have you tried v3.8?

blin08, 2017-09-14:
I just tired it on v3.8.2 with my Spektrum AR12310T reciever. The SRXL input into the pixhawk shows sporadic movement on almost all the channels on the radio calibration page.

Massild, 2018-04-09:
This problem persists.  I have noticed it in both plane and copter.  Going to plane 3.5.2 or copter 3.3.3 both resolve the issue so it was introduces immediately after those releases.

Massild, 2018-04-09:
My configuration is:
Controller: pixhawk
TX: DX8 G2
RX: AR8010T

Observed behavior:
If I am at a newer version of the software the elevator jumps from 900 to 1500 randomly. Other channels jump as well but this is one example.  This does not occur on the 3.3.3 version of copter or the 3.5.2 version of plane.