ArduPilot/ardupilot #4450: Add EK2 pos/vel addaptative params when using RTK

roque-canales, 2016-07-11:
AC3.4.1

Using RTK, it will be perfect that EK2 adapt automaticly it's param's as VEL/POS NOISE/GATES + alt source following RTK GPS status was in single / float / integer

WickedShell, 2017-09-15:
Given that the EKF looks at the GPS accuracy estimates, is there much to be gained my messing with the gates? @priseborough any opinions?

priseborough, 2017-09-15:
There are two parameter types. The *_M_NSE parameters set the minimum value of noise that the GPS fusion will be allowed to use regardless of what the receiver reports. This is there to protect against receivers that are overly optimistic. The *_I_GATE parameters set the number of percentile standard deviations allowed for the innovation before it fails consistency checks and the measurement is rejected. The size of the gate therefore adjusts to the reported accuracy of the GPS, however the minimum accuracy is always bounded by the relevant *_M_NSE parameter.

amilcarlucas, 2017-09-25:
Does that mean that *_M_NSE should be in the cm range when using RTK GPS ?

priseborough, 2017-09-26:
Depends on the receiver. I have seem some that don't perform well dynamically. In general the param should be reduced for an RTK receiver, but it should be done in steps and the innovations and innovation test ratios should be monitored. The parameters defining the IMU and GPS sensor locations should also be set to an equivalent level of accuracy.

kostkar, 2018-09-02:
May I highlight here, that especially for the altitude source this could prove to be a safety concern, the RTK fix or float height(cm) is pretty great but the SPP GPS altitude is very poor (5+m variations). Therefore, a loss of RTK fix/float, leads to a highly unsafe fallback, see also #7027.

roque-canales, 2018-10-13:
@priseborough 
You do not understand our request….
EK2_ALT_SOURCE = 2
What we ask is if the gps goes from RTK FIX to 3D Fix in flight, that EKF goes from GPS to Baro to estimate altitude.

Could you please add PR and merge it for 3.6 ?