ArduPilot/ardupilot #4459: Quadplane: Allow rudder to operate in VTOL modes

iskess, 2016-07-12:
This request was made in a [forum post](http://discuss.ardupilot.org/t/questions-and-observations-from-my-quadplane-maiden/10035/3) and I wanted to add it here for tracking. 

Question: Why does the rudder not operate in Q modes? Wouldn't it help to yaw the aircraft using the wind?

@tridge: "you are right that we could use rudder in QHOVER and QLOITER, although I don't think it would be very effective at low speeds. It is probably worth a try though. The trick will be getting the gain right. I'll see if I can come up with a reasonable approach."

iskess, 2016-07-15:
Strange thing happened today that is related to this issue. Whilst in QLOITER, the rudder went full deflection for no apparent reason. Rudder stick was not active. After switching to Manual Mode the rudder returned to center. This behavior was not repeated on the next loiter. 

[image]

BIN:
https://www.dropbox.com/s/47km8ds0nkhai9f/23.BIN?dl=0

magicrub, 2017-01-17:
sorry for not giving this issue attention. I've been a bit out of the loop for quadplane and @tridge has been super busy. We really appreciate you complaining about problems like this, we need important feedback like this.