ArduPilot/ardupilot #4468: RelAlt above Home missing in 3.4

ThorstenDP, 2016-07-14:
#### Issue details
Hi All,

in AC 3.3 RelAlt above Home was logged in the GPS data. This hase been removed in May by @tridge .
[https://github.com/ArduPilot/ardupilot/commit/7ab1367ec44bd256d419f36ad9c6854b8aacc712](https://github.com/ArduPilot/ardupilot/commit/7ab1367ec44bd256d419f36ad9c6854b8aacc712)

RelAlt above home is the auto mission altitude so it is important for several use cases. Currently it is only recorded in the CAM log. But a CAM_TRIGG is not relevant for many types of missions and as soon as CAM_TRIGG_DIST is set to 0, RelAlt above Home is not recorded.

Instead we have RelAlt above the EKF origin recorded. But there are differences of several meters to RelAlt above Home. So it is not showing the values used in Mission Planning. See [http://discuss.ardupilot.org/t/relalt-disparity/10132](http://discuss.ardupilot.org/t/relalt-disparity/10132) for some logs and discussion with @OXINARF 

Would be great if RelAlt above Home could be logged for the entire flight in AC 3.4 - maybe as an additional RelAlt value in the POS message.

Thanks a lot in advance!
Thorsten

#### Version
AC3.4

#### Platform
[?] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Logs
 [http://discuss.ardupilot.org/t/relalt-disparity/10132](http://discuss.ardupilot.org/t/relalt-disparity/10132)

WickedShell, 2017-03-10:
@ThorstenDP POS does have a RelHome altitude. And in 3.5 it even has one relative to aircraft home vs origin home.

ThorstenDP, 2017-03-10:
@WickedShell thanks for the hint about 3.5! I must have missed that... But I just started tesing 3.5. In 3.4 I guess only RelAlt above EKF origin is logged in POS, which might be off about several meters compared to RelAlt above home.

OXINARF, 2017-03-10:
I think it's not in 3.5-rc1, but it will be in 3.5-rc2 and final release.

Anyway, since this in master I'm closing this.

closed 2017-03-10 by OXINARF

rmackay9, 2017-03-10:
In case it helps, the alt above home is also in the CTUN message in all versions.

OXINARF, 2017-03-11:
@rmackay9 Are you talking about the Alt field? Because that's altitude above origin. We have BAlt but that comes directly from the baro and it's not always exactly the same.

rmackay9, 2017-03-11:
Ah, @OXINARF is absolutely right.

ThorstenDP, 2017-03-11:
@OXINARF Thanks! I wanted to test master anyway the next days. I'll make some tests.