ArduPilot/ardupilot #4471: Unable to $ modules/waf/waf-light plane

RoverHunter, 2016-07-14:
#### Issue details
I am unable to complete the tutorial done by tridge of "[Using SITL with your own GCS](http://ardupilot.org/dev/docs/sitl-with-xplane.html)".
When $ modules/waf/waf-light plane I get the following error:

/cygdrive/c/ardupilot
$ make clean
make -C ArduPlane clean
make[1]: Entering directory '/cygdrive/c/ardupilot/ArduPlane'
make[1]: Leaving directory '/cygdrive/c/ardupilot/ArduPlane'
make -C ArduCopter clean
make[1]: Entering directory '/cygdrive/c/ardupilot/ArduCopter'
make[1]: Leaving directory '/cygdrive/c/ardupilot/ArduCopter'
make -C APMrover2 clean
make[1]: Entering directory '/cygdrive/c/ardupilot/APMrover2'
make[1]: Leaving directory '/cygdrive/c/ardupilot/APMrover2'
make -C AntennaTracker clean
make[1]: Entering directory '/cygdrive/c/ardupilot/AntennaTracker'
make[1]: Leaving directory '/cygdrive/c/ardupilot/AntennaTracker'

/cygdrive/c/ardupilot
$ ./modules/waf/waf-light configure --board sitl
Setting top to                           : /cygdrive/c/ardupilot
Setting out to                           : /cygdrive/c/ardupilot/build
Setting board to                         : sitl
Checking for 'g++' (C++ compiler)        : /usr/bin/g++
Checking for 'gcc' (C compiler)          : /usr/bin/gcc
Checking for need to link with librt     : not necessary
Checking for HAVE_CMATH_ISFINITE         : yes
Checking for HAVE_CMATH_ISINF            : yes
Checking for HAVE_CMATH_ISNAN            : yes
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes
Checking for header endian.h                   : yes
Checking for header byteswap.h                 : yes
Checking for program 'python'                  : /usr/bin/python
Checking for python version >= 2.7.0           : 2.7.10
Checking for program 'git'                     : /usr/bin/git
Checking for program 'size'                    : /usr/bin/size
Benchmarks                                     : disabled
Unit tests                                     : enabled
'configure' finished successfully (3.799s)

/cygdrive/c/ardupilot
$ ./modules/waf/waf-light plane
Waf: Entering directory `/cygdrive/c/ardupilot/build/sitl'
[1/2] Running Submodule update: gtest
[2/2] Running Submodule update: mavlink
Waf: Leaving directory `/cygdrive/c/ardupilot/build/sitl'
Build failed
Traceback (most recent call last):
  File "/cygdrive/c/ardupilot/modules/waf/waflib/Task.py", line 281, in process
    ret = self.run()
  File "<string>", line 6, in f
  File "/cygdrive/c/ardupilot/modules/waf/waflib/Task.py", line 179, in get_cwd
    self.generator.bld.fatal('Working folders given to tasks must be Node objects')
  File "/cygdrive/c/ardupilot/modules/waf/waflib/Context.py", line 481, in fatal
    raise self.errors.ConfigurationError(msg, ex=ex)
ConfigurationError: Working folders given to tasks must be Node objects

Does any one have any idea how to solve this compilation problem?

#### Version
v3.6.0 - 2016-07-14 02:46:17 - f579c3e6ebc3531e8fa778cba44eefe02db13dec 

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

OXINARF, 2016-07-14:
Can you try running `git submodule update --recursive`?

Next time don't open an issue for this, you should post at the forum (http://discuss.ardupilot.org) or in Gitter. Issues list are confirmed bugs or enhancement requests.

RoverHunter, 2016-07-14:
Unfortunately that did not solve this problem:

/cygdrive/c//ardupilot
$ git submodule update --recursive
Submodule path 'modules/PX4Firmware': checked out '44574c48f97b2cb39d042789983af4376cc18478'
Submodule path 'modules/mavlink': checked out 'e0f56de31cad7a592e1bdcae88825cbb5698c79d'

/cygdrive/c//ardupilot
$ ./modules/waf/waf-light plane
Waf: Entering directory `/cygdrive/c//ardupilot/build/sitl'
[1/2] Running Submodule update: gtest
[2/2] Running Submodule update: mavlink
Waf: Leaving directory `/cygdrive/c//ardupilot/build/sitl'
Build failed
Traceback (most recent call last):
  File "/cygdrive/c//ardupilot/modules/waf/waflib/Task.py", line 281, in process
    ret = self.run()
  File "<string>", line 6, in f
  File "/cygdrive/c//ardupilot/modules/waf/waflib/Task.py", line 179, in get_cwd
    self.generator.bld.fatal('Working folders given to tasks must be Node objects')
  File "/cygdrive/c//ardupilot/modules/waf/waflib/Context.py", line 481, in fatal
    raise self.errors.ConfigurationError(msg, ex=ex)
ConfigurationError: Working folders given to tasks must be Node objects

OXINARF, 2016-07-14:
Ok, try `git submodule update --init --recursive`. We have found a bug but I think that command should fix it for now.

RoverHunter, 2016-07-14:
Thank you for your help, but that did not fix it:

/cygdrive/c/ardupilot
$ git submodule update --init --recursive
Submodule 'cmake_hexagon' (https://github.com/ATLFlight/cmake_hexagon.git) registered for path 'cmake/cmake_hexagon'
Submodule 'dspal' (https://github.com/PX4/dspal) registered for path 'dspal'
Cloning into 'cmake/cmake_hexagon'...
remote: Counting objects: 277, done.
remote: Total 277 (delta 0), reused 0 (delta 0), pack-reused 277
Receiving objects: 100% (277/277), 49.68 KiB | 0 bytes/s, done.
Resolving deltas: 100% (131/131), done.
Checking connectivity... done.
Submodule path 'modules/PX4Firmware/src/lib/DriverFramework/cmake/cmake_hexagon': checked out 'ba681fcd0d4c5ac95636dd8e9d8536dbd81beabd'
Cloning into 'dspal'...
remote: Counting objects: 748, done.
Receiving objects:  78% (remote: Total 748 (delta 0), reused 0 (delta 0), pack-reused 747
Receiving objects: 100% (748/748), 251.95 KiB | 0 bytes/s, done.
Resolving deltas: 100% (446/446), done.
Checking connectivity... done.
Submodule path 'modules/PX4Firmware/src/lib/DriverFramework/dspal': checked out '8f39699c6b79447f32f0714629da790bbf271aa5'
Submodule 'test/dspal_tester/cmake_hexagon' (https://github.com/ATLFlight/cmake_hexagon.git) registered for path 'test/dspal_tester/cmake_hexagon'
Cloning into 'test/dspal_tester/cmake_hexagon'...
remote: Counting objects: 277, done.
remote: Total 277 (delta 0), reused 0 (delta 0), pack-reused 277
Receiving objects: 100% (277/277), 49.68 KiB | 0 bytes/s, done.
Resolving deltas: 100% (131/131), done.
Checking connectivity... done.
Submodule path 'modules/PX4Firmware/src/lib/DriverFramework/dspal/test/dspal_tester/cmake_hexagon': checked out 'ba681fcd0d4c5ac95636dd8e9d8536dbd81beabd'
Submodule 'matrix' (https://github.com/PX4/Matrix) registered for path 'matrix'
Cloning into 'matrix'...
remote: Counting objects: 1060, done.
remote: Total 1060 (delta 0), reused 0 (delta 0), pack-reused 1060
Receiving objects: 100% (1060/1060), 822.58 KiB | 474.00 KiB/s, done.
Resolving deltas: 100% (749/749), done.
Checking connectivity... done.
Submodule path 'modules/PX4Firmware/src/lib/ecl/matrix': checked out '95e3d7d6ce5274d9b7fec76c0093dc5eaa381abf'
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'dsdl'
Submodule 'libuavcan/dsdl_compiler/pyuavcan' (https://github.com/UAVCAN/pyuavcan) registered for path 'libuavcan/dsdl_compiler/pyuavcan'
Cloning into 'dsdl'...
remote: Counting objects: 515, done.
remote: Total 515 (delta 0), reused 0 (delta 0), pack-reused 515
Receiving objects: 100% (515/515), 66.27 KiB | 0 bytes/s, done.
Resolving deltas: 100% (283/283), done.
Checking connectivity... done.
Submodule path 'modules/PX4Firmware/src/modules/uavcan/libuavcan/dsdl': checked out '9804a3e6972825586be252ce08dd899f44994b14'
Cloning into 'libuavcan/dsdl_compiler/pyuavcan'...
remote: Counting objects: 1062, done.
remote: Total 1062 (delta 0), reused 0 (delta 0), pack-reused 1062
Receiving objects: 100% (1062/1062), 208.14 KiB | 398.00 KiB/s, done.
Resolving deltas: 100% (718/718), done.
Checking connectivity... done.
Submodule path 'modules/PX4Firmware/src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'dsdl'
Cloning into 'dsdl'...
remote: Counting objects: 515, done.
remote: Total 515 (delta 0), reused 0 (delta 0), pack-reused 515
Receiving objects: 100% (515/515), 66.27 KiB | 0 bytes/s, done.
Resolving deltas: 100% (283/283), done.
Checking connectivity... done.
Submodule path 'modules/PX4Firmware/src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'dsdl'
Cloning into 'dsdl'...
remote: Counting objects: 515, done.
remote: Total 515 (delta 0), reused 0 (delta 0), pack-reused 515
Receiving objects: 100% (515/515), 66.27 KiB | 0 bytes/s, done.
Resolving deltas: 100% (283/283), done.
Checking connectivity... done.
Submodule path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'

/cygdrive/c/ardupilot
$ ./modules/waf/waf-light plane
Waf: Entering directory `/cygdrive/c/ardupilot/build/sitl'
[1/2] Running Submodule update: gtest
[2/2] Running Submodule update: mavlink
Waf: Leaving directory `/cygdrive/c/ardupilot/build/sitl'
Build failed
Traceback (most recent call last):
  File "/cygdrive/c/ardupilot/modules/waf/waflib/Task.py", line 281, in process
    ret = self.run()
  File "<string>", line 6, in f
  File "/cygdrive/c/ardupilot/modules/waf/waflib/Task.py", line 179, in get_cwd
    self.generator.bld.fatal('Working folders given to tasks must be Node objects')
  File "/cygdrive/c/ardupilot/modules/waf/waflib/Context.py", line 481, in fatal
    raise self.errors.ConfigurationError(msg, ex=ex)
ConfigurationError: Working folders given to tasks must be Node objects

OXINARF, 2016-07-14:
Have you looked at my edit in the previous post?

guludo, 2016-07-14:
I think the init must be called in a separate command, haven't tested though:

```bash
git submodule init
git submodule update
```

By the way, #4472 should fix this problem.

OXINARF, 2016-07-14:
@guludo That isn't needed when you run the update with --init, it does it for you

RoverHunter, 2016-07-14:
Thank you @OXINARF, apparently that solved the problem. I was finally able to build it.
Thanks.

OXINARF, 2016-07-14:
Alright, closing now, fix is in #4472 and should be in master soon.

closed 2016-07-14 by OXINARF

guludo, 2016-07-14:
> @guludo That isn't needed when you run the update with --init, it does it for you

Hm, yeah...

esamet, 2023-01-11:
> git submodule update --init --recursive

yes this is actually work with ubuntu 22.04