ArduPilot/ardupilot #4477: Quadplane bug: Qmotors activate with hardware safety switch disarmed.

iskess, 2016-07-15:
#### Issue details
If I press the hardware safety switch so that its light is flashing, and I see "SAFE" on the MP HUD, then turn off the RC TX, the Qmotors fire up and attempt an RTL. 
If I first disarm using left Rudder (Manual mode required, see #4476), then disarm using the hardware switch, then the motors will not engage when Failsafe is activated. 

The hardware switch should be a master disarm switch which does not require rudder disarming. 

#### Version
3.6

#### Platform

[ X ] QPlane

#### Airframe type
3DR Aero

#### Hardware type
Pixhawk

#### Logs
https://www.dropbox.com/s/d1g789h765phvcv/22_no%20disarm.BIN?dl=0

iskess, 2016-07-23:
Can someone please add a SAFETY label to this issue? This is quite hazardous.

tridge, 2016-08-29:
I think this was already fixed in the changes I made to safety switch handling on FMU outputs, but I have added an additional check in the quadplane code to be sure

closed 2016-08-29 by tridge