ArduPilot/ardupilot #4478: Quadplane: Allow horizontal motor to assist in forward flight in QHOVER

iskess, 2016-07-15:
#### Issue details
Q_VFWD is a great feature that allows the Quadplane to use its horizontal motor against the wind to maintain position in Q_LOITER. It seems to me that this would be a valuable feature in QHOVER too. Currently the wing must pitch down to move forward, creating a larger load on the quad motors. If reverse thrust is set up, the horizontal motor should be available to assist in either forward or reverse flight while maneuvering in QHOVER with the pitch attitude near zero. . 

#### Version
3.6

#### Platform
[ X ] QPlane

#### Airframe type
3DR Aero

#### Hardware type
Pixhawk

iskess, 2017-02-11:
Tridge, 
Is there a reason why Q_VFWD is not available for Q_HOVER? Is it by design? The more I familiarize myself with QPlane, I find that I prefer to maneuver to landing in Q_HOVER. QLoiter is really not the best for maneuvering. In a strong headwind I really miss having forward thrust to avoid pitching the nose down to penetrate. Can it at least be an option? Thank you.

tridge, 2017-02-15:
the VFWD option is based around the desired velocities in the loiter controller. That loiter controller is not running in QHOVER, so the information is not available.
It is also a safety mechanism, as QHOVER is a good fallback mode in case any problems develop with position hold. In order to do VFWD or weathervaning we need to be doing position hold.
Is there some reason you don't just use QLOITER?

tridge, 2017-02-15:
> QLoiter is really not the best for maneuvering

perhaps you just need to tune QLOITER to be more like QHOVER in handling? Have a look at the following parameters:
Q_WP_LOIT_MAXA
Q_WP_LOIT_MINA
Q_WP_LOIT_SPEED

iskess, 2017-02-15:
I will try to tune QLOITER better and that will help. One reason I don't like it is because it causes an abrupt and stressful transition from forward flight. This requires me to switch to QHOVER, then QLOITER which is a lot of messing around with switches at a critical time. I never plan to stay in one location, I only have about 1 min to get the bird on the ground before my Qbattery is dead. 
Perhaps QHOVER could have a more simplistic method of using the horizontal motor, no position controller needed. 
If I push the stick forward, rather than pitching nose down, it increases thrust on the horizontal motor. If reverse thrust is enabled, it does the same if I pull back on the stick. What do you think about that as an option?

iskess, 2017-02-15:
And weathervaning? 
If I want the plane to weathervane in QHOVER, can I just turn down a yaw parameter? I want to have yaw control when I command it, but when the stick is centered, I don't want the quad to fight the tail from weathervaning. 
Also, can we activate the rudder in Q_Modes? See #4459

iskess, 2017-05-18:
This topic was discussed here: http://discuss.ardupilot.org/t/tiltrotor-support-for-plane/8805/296
I just wanted to copy it here for tracking. 

> Tell me if my logic is flawed. QHOVER is equivalent to Copter's ALT HOLD mode, and in ALT HOLD mode forward velocity is just a byproduct of the pitch angle that is controlled the same as in STABILIZE. So the goal would be to replace Copter's pitch control with motor tilt. The priorities are: 1) maintain altitude 2) maintain level (hover) pitch, 3) increase motor tilt proportionally to pitch stick until either 1) or 2) can no longer be achieved. This will define the max forward velocity in QHOVER. 
> For a conventional Quadplane, we use the horizontal motor throttle rather than motor tilt, so we are not limited because the motor isn't also trying to handle altitude and attitude duties. In this case, I would just like to have the option to lock the pitch at level and have the horizontal motor throttle mapped to forward movement of the pitch stick.
> 
> I feel that QSTABILIZE should be the only mode where a winged aircraft is using down pitch to accelerate forward. 
> Thanks for hearing me out.
__________________________________________________________________________________

> I think this does make sense as an option, but I'm not yet convinced it should be the default. I'd like to try it out on my convergence and see what it's like.
> Cheers, Tridge

kd0aij, 2019-06-08:
@iskess https://github.com/ArduPilot/ardupilot/pull/11378 is somewhat related to this issue. It adds an RC channel for independent control of forward speed of a tiltrotor.  The same approach could be used for a non-tilt quadplane also.

IamPete1, 2021-09-12:
You can now control manually with RCx_OPTION 209 https://github.com/ArduPilot/ardupilot/pull/14520

peterbarker, 2024-07-31:
See `Q_FWD_THR_USE`

closed 2024-07-31 by peterbarker