ArduPilot/ardupilot #4481: SITL with ArduPlane v3.6.0 Unable to take off

RoverHunter, 2016-07-15:
#### Issue details
I am unable to simulate using XPlane 10.45 with Mission Planner. I have follow this Tridge tutorial of [Using SITL with your own GCS](http://ardupilot.org/dev/docs/sitl-with-xplane.html#using-sitl-with-your-own-gcs) to get everything ready for SITL but unfortunately the plane does not takeoff.
I had some trouble compiling but with the help of issue [#4471](https://github.com/ArduPilot/ardupilot/issues/4471) I was able to compile with not errors.

On Mission Planner, under Simulation tab, after Sim Link Start/Stop and connect to the simulation via TCP, I get no Plane GPS, Plane IMU, Ardupilot Output or Autopilot Status data, which is strange when comparing with older ardupilot version such as the 3.4.

I do a mission, with takeoff, arm the plane, and when in Auto, the plane does not takeoff.

When I open a new airport on XPlane, the plane is relocated correctly on Mission Planner, and I can see under Net Connections / Data that the last message was received at 0.0 seconds ago, so I assume Mission Planner and Xplane are talking to each other correctly.

What could I be doing wrong?

#### Version
Using fc41e0e44e5c28ffbd2881cbad825b7b1736b2e0 build with Mission Planner 1.3.39.1 build 1.1.6040.17593.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

lucasdemarchi, 2016-07-15:
I don't have a windows machine handy so no way to test the xplane part. What I did was with plane instead and mavproxy:

```
Tools/autotest/sim_vehicle.py --vehicle ArduPlane --map

MANUAL> wp load Tools/autotest/CMAC-circuit.txt
MANUAL> mode auto 
AUTO> arm throttle
```

and it succesfully takes off.  So maybe it's:
  - something new that got broken on windows or
  - with xplane model or 
  - with the sequence of commands you are doing

@tridge @meee1  could you help?

RoverHunter, 2016-07-20:
Can anyone confirm the problem with XPlane?

khancyr, 2018-04-24:
Xplane and SITL have been intensively update since this problem. I am closing as it is outdated and surely not relevant anymore. Feel free to open a new post on discuss in case of problem

closed 2018-04-24 by khancyr

moooeeeep, 2020-08-31:
Just in case someone else is having the same problem...

Probably the issue was caused by OP trying to takeoff using similar steps as with a copter (arm throttle, takeoff <altitude>).

In order for the plane to takeoff it needs to have specified another waypoint in addition (e.g., a loiter_unlim wp to go to after takeoff).

The documentation could be clearer about this. AFAIK it gives no reason what it is about to load that waypoint file. When I find the relevant page next time, maybe I'll try to add this.