ArduPilot/ardupilot #4486: "__builtin_bswap16 was not declear in this scope" on windows toolchain

EwingKang, 2016-07-16:
#### Issue details
Hello, everyone
Today when I'm trying to teach someone to compile the code, I ran into this problem
This update in [bytespap.h](https://github.com/ArduPilot/ardupilot/blob/b9f65996bcc5fd7a8b0a0d7146bebefd745ab4a6/libraries/AP_Common/missing/byteswap.h) seems to break the px4 toolchain on windows.
 cxx error about missing __builtin_bswap16

> missing/byteswap.h:11:42: errror: '__builtin_bswap16' was not declared in this scope compilation terminated` due to -Wfatal-errors. 

I'm currently using windows toolchain v14 as stated on the [official website](http://ardupilot.org/dev/docs/building-px4-with-make.html).

#### [commit acaf0c6](https://github.com/ArduPilot/ardupilot/commit/acaf0c6a5d5a111ad667067c7276c6deae0f4ac9)

#### Platform
[  ] Plane

#### Hardware type
pixhawk

Someone says on the[ stackoverflow ](http://stackoverflow.com/questions/18208719/c-pre-processor-detect-if-a-builtin-function-is-available)that it's because of the outdated gcc. I'll try to update the gcc of my mingw32.
Am I doing the right thing?
I'm not really familiar with programming process, guessing this is sort of not directly related to the code of ardupilot, but I'm not sure where should I go. Sorry for the interruption and thanks in advance for any advice or help.

EwingKang, 2016-07-16:
Okay, according to [this bug](https://github.com/PX4/Firmware/issues/833)I've also check the version of arm-none-eabi-gcc by  

> arm-none-eabi-gcc --version 

I tried to update it by downloading the new version zip for win32, exract file, and brutally replace everything with the same name in c:/px4/toolchain/. 
The first attempt of update to latest version 5.4 failed and a make warning says I sould do 4.9.3, so I download the [4.9.2015-q3](https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update) and it actually worked! 
This is probably not a very nice way to update software, and I didn't test the resulting .px4 file. But at least the compile passed.

OXINARF, 2016-07-16:
This was fixed with using the compiler minimum supported version. Closing.

closed 2016-07-16 by OXINARF

EwingKang, 2016-07-17:
Pardon me, I'm not quite get what you means by "using the compiler minimum supported version". Does that means now the make will check the version of "arm-none-eabi-gcc" automatically? I just clone the newest master, the (before manual updated) compiler still gives the error 2 message. Or should I download the toolchain again?

OXINARF, 2016-07-17:
No, it means that if you use the 4.9 version you won't have a problem. That is what we use nowadays. We are looking at a way to provide an updated Windows installer for that version.

EwingKang, 2016-07-18:
Okay, got it. Thanks for your attention and reply!

EwingKang, 2016-07-28:
Update: Newest windows installer should fix the issue at:
http://firmware.ardupilot.org/Tools/PX4-tools/

futurestark, 2016-10-13:
sorry that in this tipic but, its question after build, 
I have build arducopter in eclipse at windows successfully the next step is how can I debug code with breakpoints in real pixhawk due to connection with stlink-v2 or jtag jlink8
Pins at board i found so how to??  
Its very actual for me help pleeeease...

magicrub, 2016-10-13:
Have you checked the wiki? Ardupilot.org. There is a section on using gdb for debugging.

magicrub, 2016-10-13:
And please start a new thread instead of hijacking  one that has been closed for many months. Your question is asking how to do something, that is best asked in the forums at discuss.ardupilot.org