ArduPilot/ardupilot #4487: ArduCopter: External custom sensor integration

volodink, 2016-07-16:
Hi. 

Recently getted into Ardupilot community and was suprised that there is no external sensors integration manual or tutorial for ArduCopter (2.6+).

I already succesfully build firmware from source in provided vagrant vm.

So in my case i need to communicate with external device via i2c or uart, get the data and send it via telemetry to the ground station. 

May be i missed something, can yo help me with some information or example?

Thanks. 

Konstantin.

magicrub, 2016-07-16:
Arducopter 2.6? Are you using the old Arduino based apm 2.6 hardware?

volodink, 2016-07-16:
@magicrub 
i use this board: [https://i.ytimg.com/vi/1V4GkwqxoNY/hqdefault.jpg](url)

hiro2233, 2016-07-17:
Hi @volodink, maybe you know, Ardupilot development for 8bits hardware are no yet supported anymore. Right now, development are with 32Bits hardwares like DSP, SoC processors systems running under NuttX, Linux and others OS. But if you want integrate APM 2.6 with AVR 8bits microcontrollers and how it's work you can see the OLD branch at this point:

https://github.com/ArduPilot/ardupilot/tree/master-AVR

Sensors integration are defined in HAL Libraries.

Maybe this PR can help you, https://github.com/ArduPilot/ardupilot/pull/3578

And for you consideration, the issues list is for bugs and feature requests for current development. This is a (development) question and those should be posted at the forum (http://discuss.ardupilot.org) or in Ardupilot Gitter channel (https://gitter.im/ArduPilot/ardupilot).

magicrub, 2016-07-17:
I'm sad that people still buy those, clones or not. Makes people think
that's the latest and greatest and they don't realize that we depreciated
it 2 years ago.

hiro2233, 2016-07-17:
@magicrub It's the transition for the world until understand everybody that 32 Bits hardware is the new one and the latest hardware, the same case happened to me.

volodink, 2016-07-17:
So, if i undestant right, there is no dev guide on this topic for even latest MEGA2560 based firmware (3.2.x)?

hiro2233, 2016-09-08:
Hi @volodink please post it on http://discuss.ardupilot.org/ maybe you will get help.
Issues tickets are for APM development.

@OXINARF  please close this.

closed 2016-09-08 by volodink