ArduPilot/ardupilot #4488: Help with Rover - GUIDED mode not responding to setpoint_position with mavros

botjunkie, 2016-07-16:
Hi guys,

I need to do use the GUIDED mode on APMRover using mavros. Basically, I want to give GPS co-ordinates of a location to the rover (through a script, not AUTO mode) and make the rover and go and stay there for a while, and then make it go to another location. I have done a similar script for COPTER and it works just fine. I have tested that out in SITL and on the real platform. However, when I set > setpoint_position (SET_POSITION_TARGET_LOCAL_NED) from the script (via mavros) the ROVER does not respond.

I am able to put the ROVER in guided mode and arming it before doing anything else. But setpoint_position just doesnt work.

I have also tried using mavproxy and it works just fine. I can change the mode to GUIDED, arm the ROVER and then click on a location on the map and it goes there. I can also override RC and have direct PWM control of throttle and steering. Having said that, I dont want to control PWM from my script. Instead I want to give the autopilot a GPS location and make it figure out the PWM commands needed to get there.

I have checked some of the related topics and tried changing parameters.
ARMING_CHECK 0
ARMING_REQUIRE 0
SYSID_MYGCS 1
But the rover still wont respond to the script.

Can you please help?

I hope I made it clear enough.

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

gmorph, 2016-07-23:
SET_POSITION_TARGET_LOCAL_NED is only implemented in Copter so that's why its not working in Rover.
GUIDED mode works by putting the vehicle into GUIDED mode and then sending a MAV_CMD_GUIDED_WP.
Please ask further questions for help in the discussion forum
http://discuss.ardupilot.org/c/ardurover

Thanks, Grant.

rmackay9, 2016-07-23:
Sorry to be annoying.  I changed the title so that it's a request to add support for that message.

khancyr, 2016-12-21:
Position setpoint are working now, see https://github.com/ArduPilot/ardupilot/pull/4716. Closing

closed 2016-12-21 by khancyr