ArduPilot/ardupilot #4491: Plane: Add wind failsafe

WickedShell, 2016-07-17:
#### Issue details
Failsafe should activate when the wind is detected to be above a set threshold for > 5 seconds. This is particularly of relevance for any time during loss of link  operations, and adds a way to help ensure the aircraft is safely returned.

#### Version
Master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

magicrub, 2016-07-18:
Are you already working on this? It's on my todo list but I haven't started.

WickedShell, 2016-07-18:
Not yet it's high on my to-do list but over a week before I'm even back at
a compiler

Naterater, 2019-01-04:
This is still probably a good feature, but because the wind estimate is so slow to respond, I would recommend at least 30 seconds.

IamPete1, 2021-08-08:
We have a scripting example for this.
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/plane-wind-failsafe.lua

peterbarker, 2024-07-31:
Closing this one; the LUA script @IamPete1 can provide this functionality in a much more flexible way than we could easily put in the C++ code.

closed 2024-07-31 by peterbarker