ArduPilot/ardupilot #4493: AP_Math : implicit conversion from 'float' to 'double'

khancyr, 2016-07-18:
Coding question.
In AP_Math we have plenty of implicit conversion from 'float' to 'double'

<script src="https://gist.github.com/khancyr/7e28596d6b79ca7ed1f58c9a1c512441.js"></script>

instead of using float constant, could we use something like 
template <class T>
float safe_asin(const T v)
{
    if (isnan(static_cast<float>(v))) {
        return 0.0f;
    }
    if (v >= static_cast<T>(1.0)) {
        return static_cast<float>(M_PI_2);
    }
    if (v <= static_cast<T>(-1.0)) {
        return static_cast<float>(-M_PI_2);
    }
    return asinf(static_cast<float>(v));
}

And doing compile time cast to prevent implicit conversion ?

lucasdemarchi, 2016-07-18:
Things to check when changing these:
  - it retains the current behavior for linux boards
  - it doesn't increase the flash size for px4

I thought the implicit conversions were not inside AP_Math itself, but rather from the callers.

khancyr, 2016-07-18:
Hum with the right formating it is better....

the errors come from double template like template float safe_asin<double>(const double v);

I am not use to template, that why I ask for help ! normally with static_cast, the constant should be cast to the right type on compile time

lucasdemarchi, 2016-07-18:
Maybe it would be better to have a float implementation and wrapper method in place through template:

```c++
float safe_asin(const float v)
{
    // implementation
}

template <class T>
float safe_asin(const T v) { return safe_asin(static_cast<float> v)); }
```

We need to do the changes and then check the things I mentioned before. Also, this part doesn't make much sense:

```
static_cast<T>(-1.0)
```

We actually want to implement the function using floats since this is what stm32 supports in hw. So, we shouldn't cast the constant to the type of the variable, but rather cast the variable to float.  I think it's easier with the wrapper method as I said.

closed 2016-11-09 by khancyr

magicrub, 2016-11-09:
Please provide a reason for closing.

khancyr, 2016-11-10:
those errors have been solve (don't remember the pr)