ArduPilot/ardupilot #4498: Support for: (#104) VICON_POSITION_ESTIMATE, MAVlink

tuloski, 2016-07-19:
#### Platform
[  ] Copter

#### Hardware type
Pixhawk

It would be nice to have support for this message to override EFK position estimation, when a ground truth is available.

Also supporting VISION_SPEED_ESTIMATE ( #103 ) and VISION_POSITION_ESTIMATE ( #102 ) would be very nice.

rmackay9, 2016-07-19:
Yes, that would be a great addition.

We do have support for a "fake gps" in that if you set the GPS_TYPE to "14" (aka "GPS_TYPE_MAV") and then send in the estimated positions using the GPS_INPUT mavlink message it will consume them just like a GPS.

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.h#L72
http://mavlink.org/messages/common#GPS_INPUT

I know it's not exactly what you're looking for but I wanted to provide this possible shorter-term work around.

tuloski, 2016-07-19:
Thanks for your answer.
GPS is a nice workaround but the problem is that is passes trough EKF. You can tune it to weight more "GPS" but still it will be filtered.

icyore, 2016-11-01:
@rmackay9 
I send the fakeGPS by mavlink. As the plugin of mavros_extra, I can't not receive any GPS data on the pixhawk and the LED don't turn green. Any other parameter should I set?

mavlink_message_t msg;
mavlink_msg_gps_input_pack_chan(
UAS_PACK_CHAN(uas),
&msg,
ros::Time::now().toNSec()/1000,
0,
32767,
0,
0,
2,
int32((lat_rad * 180 /M_PI)*10000000),
int32((lon_rad * 180 /M_PI)*10000000),
float(alt),
0.5,
1.0,
0.0,
0.0,
0.0,
0.0,
0.0,
0.0,
11);
UAS_FCU(uas) -> send_message(&msg);

ardupilot ArduCopter 3.4.0 rc 7 quadrotor
pixhawk
ttyUSB0 connect to the telem2

peterbarker, 2020-06-19:
We consume these messages now.

closed 2020-06-19 by peterbarker