ArduPilot/ardupilot #4499: Unnedeed code in  AntennaTracker binary

night-ghost, 2016-07-19:
#### Issue details
Size of AntennaTracker binary is more than it can be because it includes useless modules:
```
  text    data     bss     dec     hex filename
    996      19       0    1015     3f7 /tmp/AntennaTracker.build/libraries/AP_Mount/AP_Mount_SToRM32_serial.o
   1402       0       1    1403     57b /tmp/AntennaTracker.build/libraries/AP_Mount/AP_Mount_Servo.o
    844       0       0     844     34c /tmp/AntennaTracker.build/libraries/AP_Mount/AP_Mount_SoloGimbal.o
   1455       0       0    1455     5af /tmp/AntennaTracker.build/libraries/AP_Mount/SoloGimbal_Parameters.o
   3524       0      12    3536     dd0 /tmp/AntennaTracker.build/libraries/AP_Mount/SoloGimbal.o
   1333       0       0    1333     535 /tmp/AntennaTracker.build/libraries/AP_Mount/AP_Mount.o
    682       0       0     682     2aa /tmp/AntennaTracker.build/libraries/AP_Mount/AP_Mount_SToRM32.o
   1232       0       0    1232     4d0 /tmp/AntennaTracker.build/libraries/AP_Mount/AP_Mount_Alexmos.o
  17545       0      24   17569    44a1 /tmp/AntennaTracker.build/libraries/AP_Mount/SoloGimbalEKF.o
   1362       0       0    1362     552 /tmp/AntennaTracker.build/libraries/AP_Mount/AP_Mount_Backend.o
    176       0       0     176      b0 /tmp/AntennaTracker.build/libraries/AP_RangeFinder/AP_RangeFinder_MAVLink.o
      364       0       0     364     16c /tmp/AntennaTracker.build/libraries/AP_RangeFinder/AP_RangeFinder_LightWareSerial.o
      312       0       0     312     138 /tmp/AntennaTracker.build/libraries/AP_RangeFinder/AP_RangeFinder_MaxsonarI2CXL.o
    122       0       0     122      7a /tmp/AntennaTracker.build/libraries/AP_RangeFinder/RangeFinder_Backend.o
    466       0       0     466     1d2 /tmp/AntennaTracker.build/libraries/AP_RangeFinder/AP_RangeFinder_analog.o
    324       0       0     324     144 /tmp/AntennaTracker.build/libraries/AP_RangeFinder/AP_RangeFinder_PulsedLightLRF.o
    256       0       0     256     100 /tmp/AntennaTracker.build/libraries/AP_RangeFinder/AP_RangeFinder_LightWareI2C.o
     1581       0       0    1581     62d /tmp/AntennaTracker.build/libraries/AP_RangeFinder/RangeFinder.o
    261       0       0     261     105 /tmp/AntennaTracker.build/libraries/AP_OpticalFlow/OpticalFlow.o
     28       0       0      28      1c /tmp/AntennaTracker.build/libraries/AP_OpticalFlow/OpticalFlow_backend.o
     30       0       0      30      1e /tmp/AntennaTracker.build/libraries/AP_OpticalFlow/AP_OpticalFlow_HIL.o
    832       0       0     832     340 /tmp/AntennaTracker.build/libraries/AP_Airspeed/AP_Airspeed.o
    100       0       0     100      64 /tmp/AntennaTracker.build/libraries/AP_Airspeed/AP_Airspeed_analog.o
   1138       0       0    1138     472 /tmp/AntennaTracker.build/libraries/AP_Airspeed/Airspeed_Calibration.o
    468       0       0     468     1d4 /tmp/AntennaTracker.build/libraries/AP_Airspeed/AP_Airspeed_I2C.o
etc.

```
what sense in the control of the gimbal or the air speed sensor - on ground? The build system should not collect all of the code but only really used.

#### Platform
[  ] All
[x] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

gmorph, 2016-07-20:
You are correct of course.  It would be great if you could submit a PR night-ghost to update the build system and resolve this issue.
Thanks, Grant.

rmackay9, 2016-07-20:
An interesting list.  No plans to attach a lidar to an antenna tracker? :-)
A slight waste of flash space but having the code there doesn't do any harm at least.

night-ghost, 2016-07-20:
Yes this is not a problem on 2MB Pixhawk but my tracker is based on FlyMaple with 103RET6 and only 512KBytes so I  just had to start optimizing code size to cram it :) (Yes, I do not want to spend another $65)

There is make.inc in AntennaTracker folder so I just comment out lines with unneeded libs - AP_Terrain, AP_Airspeed, AP_OpticalFlow and AP_Mount. And viola:

Final Size:
   text    data     bss     dec     hex filename
 499788    4624    9032  513444   7d5a4 /tmp/AntennaTracker.build/AntennaTracker.elf

Excluding of AP_RangeFinder gives link error because of

AP_AHRS_NavEKF(AP_InertialSensor &ins, AP_Baro &baro, AP_GPS &gps, RangeFinder &rng, NavEKF &_EKF)

 but was able to fit and with it, so I do not investigate the way to remove it.