ArduPilot/ardupilot #4503: Need capabilities for HOME_POSITION support (all messages)

hamishwillee, 2016-07-20:
#### Issue details

Historically the home position was stored in waypoint_0. AC3.3 introduced support for the HOME_POSITION and MAV_CMD_GET_HOME_POSITION. AC3.4 added support for MAV_CMD_SET_HOME_POSITION.

Dronekit and others need an easy way to determine when home position messages are supported so that code to get the information from waypoints can be disabled.  The best way is probably to add capabilities. We probably actually need two capabilities (one for get and one for set) since these appear in different versions.

#### Version
3.3, 3.4

#### Platform
[ X ] All except AntennaTracker
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
All

#### Hardware type
All

#### Logs
NA

hamishwillee, 2016-07-20:
PS FYI, I am not certain HOME_POSITION does actually get sent on Copter 3.3 - certainly the DK-SITL version does not support this message (but Copter 3.4 does).

IamPete1, 2021-08-30:
Since we never did this at the time I suspect the need for the capability reporting has passed.

IamPete1, 2021-09-11:
Please re-open if this is still a issue.

closed 2021-09-11 by IamPete1