ArduPilot/ardupilot #4509: feature request: elevator's trim value changes depends on velocity changes

nocternal, 2016-07-20:
#### Issue details
We know at different velocity to fly level needs different elevator servo value. How about offer some advanced parameters to do this? It's kind of auto trim during plane's fly.For example, when this function enabled , we can let user set three different reference velocitys and different elevator servo values.These values need to have a different speed level flight test to get. Then let a interpolation function work, if airspeed within this setted velocity range, then do a interpolation for flight speed and reference speed. If out of maximum or minimum speed range,then the elevator trim value is also limited at user setted maxmum and minimum value.
How about this feature?Any advice?  

#### Platform
[  ] Plane

magicrub, 2016-07-20:
This is already handled for all speeds by the integrator. What's wrong with
the current scheme?

magicrub, 2017-06-12:
I think this is implemented already so closing. Plus, there's a new adaptive controller in the pipeline that will make the current behavior even better

closed 2017-06-12 by magicrub